ros2/ros2 #275: ros2 debian packages

firesurfer, 2016-09-27:
I managed to create set of debian packages with checkinstall that install a ros2 workspace to /opt/ros2_ws.
Would it be ok if I provide this set of packages on my own webspace ?

dirk-thomas, 2016-09-27:
Since the code is licensed under Apache 2 you can do whatever you like :wink: Usually we just ask to add a note to clarify that these are package you created that nobody expects them to be "official" packages.

firesurfer, 2016-09-27:
Ok,
I just added a license.txt and a readme.txt that states that these aren't official packages.
http://firesurfer.de/ros2/

dirk-thomas, 2016-09-27:
You might want to mention which rmw implementations your packages are built against.

firesurfer, 2016-09-27:
Thanks! I just added this information.

dirk-thomas, 2016-09-27:
If you only built `FastRTPS` you might want to exclude the Connext and OpenSplice specfic packages (since they won't do anything).

closed 2016-10-14 by dirk-thomas