ros2/ros2 #279: [windows] update install instruction wiki

Karsten1987, 2016-09-29:
The install instructions for the windows binary installation should be updated for Opensplice and Opencv requirements. See https://github.com/ros2/ros2/issues/278

Karsten1987, 2016-09-29:
can I simply edi the wiki page or do we have any PR-like procedure for it?

dirk-thomas, 2016-09-29:
You can go ahead and edit the page. It would be good to post the link to the diff here afterwards. Also @wjwwood is watching all changes and will review them "after the fact".

mikaelarguedas, 2016-09-29:
no we tend to edit directly the pages for wiki changes there is no review
process invovled

mikaelarguedas, 2016-09-29:
it would also be great to update the [Tutorials page](https://github.com/ros2/ros2/wiki/Quality-Of-Service) to explain the new `-b` (aka burger) option so that people can try it without to hook a webcam. It can be done later and referenced on another ticket though

dhood, 2016-09-29:
could you address https://github.com/ros2/ros2/issues/269 too while you're there

closed 2016-10-18 by Karsten1987