ros2/ros2 #281: Beta 1 on the 16th of December

mikaelarguedas, 2016-10-18:
* [ ] Composition @wjwwood @dirk-thomas 
   * [ ] Using class loader, pluginlib, C++ only with intention that interface is less language specific where possible
 * [ ] Lifecycle management  @Karsten1987 @tfoote  
 * [ ] QoS demos: unreliable comms, illustrated by wifi out-and-back @dhood 
 * [ ] Design documents for open questions/discussions @wjwwood (and everyone else)
  * [ ] Node-to-participant mapping design (https://github.com/ros2/design/pull/90)
  * [ ] Uniqueness of node names design
  * [ ] Separator namespace
  * [ ] Async vs sync publishing
  * [ ] Service provider: content-oriented or host-oriented
  * [ ] How to communicate remappings (cmd args, launch file, param file, etc.)

**Stretch goals:**
 * [ ] Creating coherent list for external contributors of separable self-contained non-critical path items
 * [ ] `rostopic list`, `rostopic echo` and friends @mikaelarguedas and @vmayoral
 * [ ] Setting parameters
   * [ ] C parameters
   * [ ] Python services + parameters
 * [ ] Bulk deb packages: core and demos (which require opencv etc

gbiggs, 2016-10-19:
I have some time to work on rostopic, if that's helpful. I spoke with @wjwwood at ROSCon and he asked me to start a topic on Discourse to discuss the options for implementing it, in light of the distributed nature of ROS2.

I can also contribute to the life cycle stuff, although it sounds like that's already covered by people with time to do it.

vmayoral, 2016-10-19:
@mikaelarguedas refer to http://discourse.ros.org/t/rostopic-implementation-discussion/674/2?u=vmayoral.

dirk-thomas, 2016-12-20:
:ship: https://github.com/ros2/ros2/releases/tag/release-beta1

closed 2016-12-20 by dirk-thomas