ros2/ros2 #284: use message/service files in the same project

sukha-cn, 2016-11-02:
In a demo, i want to use custom message in the same project like this:
`
#include "my_test/srv/empty.hpp"
...
this->create_service<my_test::srv::Empty>("my_test_server", [](){

});
...
`
Then how to write CMakeList.txt?

Currently my CMakeList.txt is:
`
cmake_minimum_required(VERSION 3.5)

project(my_test)

if(NOT WIN32)
  set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11 -Wall -Wextra")
endif()

find_package(ament_cmake REQUIRED)
find_package(rclcpp REQUIRED)
find_package(rosidl_default_generators REQUIRED)
find_package(rmw_implementation_cmake REQUIRED)

#include_directories(include)

rosidl_generate_interfaces(${PROJECT_NAME}
  "srv/Empty.srv"
)

ament_export_dependencies(rosidl_default_runtime)
add_executable(my_test_node
  src/my_test.cpp
)
ament_target_dependencies(my_test_node
  "rclcpp"
)
ament_package()
`

dirk-thomas, 2016-11-02:
There are some packages like [test_communication](https://github.com/ros2/system_tests/tree/master/test_communication) which contain messages and also use them within the same project file.

Most importantly you need a dependency from your executable target to the rosidl code generation target: https://github.com/ros2/system_tests/blob/07aa8fa2bf62ddefebf4c50f9a8c8348a6a860fb/test_communication/CMakeLists.txt#L59

Additionally you need to add the necessary include directories etc. to the executable target so that it finds the generated headers: https://github.com/ros2/system_tests/blob/07aa8fa2bf62ddefebf4c50f9a8c8348a6a860fb/test_communication/CMakeLists.txt#L61-L62

sukha-cn, 2016-11-03:
Thank you for answering. It works now.
I think it is necessary to add a macro or function about this in ament.

dirk-thomas, 2016-11-03:
The custom function `rosidl_target_interfaces` provided by the `rosidl_cmake` package could actually add the target dependency too. I think that is already the sleekest API for a user to use - its a single function call then.

I assume you meant that with your request "to add a macro or function about this in ament"? In general `ament` doesn't know anything about messages and therefore can't provide any such functionality. But the `rosidl_cmake` package does exactly that - provide a CMake function to make it easy for the user.

dirk-thomas, 2016-11-03:
I just saw your new repo [turtlesim-ros2](https://github.com/sukha-cn/turtlesim-ros2). If I may make two suggestions:

* It would be good to base the work on the latest version from the `kinetic-devel` branch since that already uses Qt 5 (and not Qt 4 anymore which is EOL)
* It would be good to make the necessary changes more visible - currently its hard to see how much modifications where necessary since the first commit contains the original code including all the modifications you had to do. E.g. by forking the repo and then applying the changes necessary to make it work with ROS 2 on a new feature branch the changes would be more visible.

dirk-thomas, 2019-09-05:
Closing since we don't plan to add any convenience functions for this use case since it is already fairly easy to do manually.

closed 2019-09-05 by dirk-thomas

SasaKuruppuarachchi, 2024-12-17:
Adding this macro to the CMakelist fixes the issue. (Humble and newer)
assuming I have a node executable named my_node

```
rosidl_get_typesupport_target(cpp_typesupport_target ${PROJECT_NAME} "rosidl_typesupport_cpp")
target_link_libraries(my_node "${cpp_typesupport_target}") 
```

[Referance](https://robotics.stackexchange.com/questions/23171/creating-a-custom-msg-in-ros2)