ros2/ros2 #285: Parameter Client directed at self actually affects the whole graph

SecretaryBirds, 2016-11-10:
I noticed some unexpected behavior with the ParameterService/ParameterClient feature.
I start up 2 nodes with different functions.
Then if I execute this code [issue.txt](https://github.com/ros2/ros2/files/583764/issue.txt) in a 3rd node, every one of those parameters shows up in the previous 2 nodes. This is problematic if you have common parameter names, such as frame_id.

wjwwood, 2016-11-29:
Sorry we haven't gotten back to you on this, we'll look into it asap.

dirk-thomas, 2019-09-05:
Due to inactivity and significant changes to parameters since then the ticket is being closed. Please feel free to comment on it if it is still a problem and the ticket can be reopened.

dirk-thomas, 2019-09-05:
Maybe related ros2/rclcpp#746.

closed 2019-09-05 by dirk-thomas