ros2/ros2 #289: enable all messages to test build time implications

codebot, 2016-11-23:


codebot, 2016-11-23:
I think this will require changes to more than one repo (?), so I put the ticket here. Feel free to move if there is a better home.

mikaelarguedas, 2016-11-23:
Comparison of build times if all messages enabled (using windows jobs results because they ran on the same hardware, jobs links: [current master](http://ci.ros2.org/job/ci_windows/1896/) and [with all messages](http://ci.ros2.org/job/ci_windows/1897/))
build only:
master: 13min51s
all messages: 19min41s
==> **+5min50s +40%**
build + test:
master: 33min44sec
all messages: 43min13sec
==> **+9min29s +30%**

Note same jobs have been ran on Linux([current master](http://ci.ros2.org/job/ci_linux/1949/) and [with all messages](http://ci.ros2.org/job/ci_linux/1950/)) and OSX([current master](http://ci.ros2.org/job/ci_osx/1501/) and [with all messages](http://ci.ros2.org/job/ci_osx/1502/)) but ran on different nodes so the time comparison may be less relevant

mikaelarguedas, 2016-11-23:
As a result I think @tfoote suggestion of:
 - uncommenting them all on master
 - applying a patch (commenting out the unused ones) before the builds

seems to be the best approach. 
The aformentionned patch could also ignore the messages used only for turtlebot demos (see https://github.com/ros2/common_interfaces/pull/19#issuecomment-258320001)
Good news is reenabling them all didn't introduce any test failure.

codebot, 2016-11-23:
Thanks for gathering this data. I think switching to building all messages is worth it for now, and we can do the patch for reduced-builds later if/when the additional few minutes is driving us crazy.

dhood, 2016-11-23:
the tests for these packages are just linters right?

if testing adds an additional 4 minutes, it would be good to get https://github.com/ros2/ci/pull/9 merged before we enable them on nightlies so that we don't waste 20*4 minutes testing them repeatedly

dhood, 2016-11-23:
as an alternative to commenting out the unused ones with a patch, we could add the packages to the list of 'blacklisted packages' for the CI jobs https://github.com/ros2/ci/blob/f325a6d3335a912b7c4ba60794faeb8d783d5082/ros2_batch_job/__main__.py#L434
less-fine control (we would still need to build everything in the `std_msgs` and `sensor_msgs` packages), but easier. I agree with @codebot that we don't have to do it now

closed 2016-11-29 by mikaelarguedas