ros2/ros2 #290: keeping track of disabled tests

mikaelarguedas, 2016-11-24:
We have a lot of disabled tests that we sometimes forget to reenable when feature is implemented. (e.g https://github.com/ros2/examples/pull/131 or https://github.com/ros2/demos/pull/83 thanks @dhood for flushing them out)
The goal here is to mark them all as skipped so that they show up as such in CI:
- [ ] expand `ament_add_*test` macros to take a skip flag
- [ ] generate result files so that they show up in Jenkins
- [ ] make ament test result report skipped tests
- [ ] go through the codebase to flag all the disabled tests as `skipped`

mikaelarguedas, 2016-11-24:
Rationale:
- provide a clear list of all the tests that we don't expect to pass
- give visibility of the features/configurations that are not expected to work in the current state
- provide insight for external contributors on where to look at if they want to improve the feature parity (xPlatforms/xRMWs/xClientLibraries)

mikaelarguedas, 2017-02-07:
@dhood assigning this to you given that you took over labeling the skipped tests

dhood, 2017-02-16:
I have gone through the codebase and the only remaining unlabeled skipped tests I could find have been labeled in https://github.com/ros2/demos/pull/114 and https://github.com/ros2/rcl/pull/109

For documentation purposes: the skipped tests are not reported as an enumerated list in the test report, but can be seen if you click through test results with >0 skipped tests, e.g.: http://ci.ros2.org/view/nightly/job/nightly_win_rel/338/testReport/(root)/test_communication/

closed 2017-02-16 by dhood