ros2/ros2 #291: Status of ignore_local_publications option for create_subscription

dhood, 2016-12-07:
I came across the [`ignore_local_publications` flag](https://github.com/ros2/rcl/blob/66208db876bba344a6733f4aa33451a028085612/rcl/include/rcl/subscription.h#L43) for `create_subscription` for the first time today. I saw this conversation https://github.com/ros2/rmw/pull/17 which makes it seem like the flag was added for the ROS 1-2 bridge.

However, the dynamic bridge [also has some logic](https://github.com/ros2/ros1_bridge/blob/cc1479b7b832c18d18a4a93523ea04deee54418b/src/dynamic_bridge.cpp#L380) for ignoring publications from itself. 

Other than some parts in the bridge, I can't find anywhere else in the ros2 codebase where the flag is ever set to true. Even in [the tests for pub-sub](https://github.com/ros2/system_tests/blob/5456c9ba5ccf1d98b626768b9ee38219fd7d0d22/test_communication/test/test_publisher_subscriber.cpp#L103) in a single process.

This is what prompts me to ask: when should this be used? Of course I feel that the answer is when you want to ignore publications from within your process, but since I couldn't find any use cases other than the bridge, and the bridge has additional logic, the answer might not be that simple.

Since we don't seem to test this functionality, I'm wondering if Fast RTPS does not behave as we expect with this setting and that's what's causing https://github.com/ros2/ros1_bridge/issues/43

wjwwood, 2016-12-08:
If you follow it down the stack, it is being used. For example in Connext static:

https://github.com/ros2/rmw_connext/blob/master/rosidl_typesupport_connext_cpp/resource/msg__type_support.cpp.em#L280-L294

It may not be implemented correctly in fastrtps, but I didn't look into that.

dhood, 2016-12-08:
Thanks for finding that link for me @wjwwood

I didn't find an equivalent in `rosidl_typesupport_introspection_cpp`, and in `rmw_fastrtps_cpp` it doesn't look it's used: https://github.com/ros2/rmw_fastrtps/blob/075bc487bee6c9a6fb6edf8f963177924040613c/rmw_fastrtps_cpp/src/functions.cpp#L1013 so that indeed might explain the difference in behaviour between the `rmw`s in https://github.com/ros2/ros1_bridge/issues/43

Sounds like the outcome here is: 
1. Improve visibility that `ignore_local_publications` isn't supported for fastrtps
2. Add tests for this functionality (the tests are what I was looking for in order to infer the expected use case)

dhood, 2016-12-08:
Question closed in favour of:
https://github.com/ros2/ros2/issues/293 +
https://github.com/ros2/rmw_fastrtps/issues/76

closed 2016-12-08 by dhood