ros2/ros2 #294: Transition to C++14

codebot, 2016-12-09:
Following the discussion on Discourse, this ticket links to branches on the repositories affected by switching to C++14. Almost all of these changes are just changing the flags from `--std=c++11` to `std=c++14`. The Windows flags were not affected because VS2015 supports C++14 by default.

CI results look clean:
 * [Linux](http://ci.ros2.org/job/ci_linux/2016/) has a few failures but these are on unrelated packages and are in the process of being fixed independently.
 * [OS X](http://ci.ros2.org/job/ci_osx/1542/)
 * [Windows](http://ci.ros2.org/job/ci_windows/1995/)

Each of repo has a `cpp14` branch that will need to be merged in.

dirk-thomas, 2016-12-09:
You mentioned above that:

> Almost all of these changes are just changing the flags from `--std=c++11` to `std=c++14`.

For easy review the only PR which is non-trivial seems to be ros2/examples#147.

I think it would be good to either see a green build on all three platforms or a build from master which has the same warnings / failing tests. Also can I assume you have tried to build the bridge and it didn't result in any warnings (since the ROS 1 code might behave differently). With that being said +1.

codebot, 2016-12-09:
Thanks for the reminder about ros1 bridge. I forgot that doesn't build by default. Anyway I just built it locally on my machine and it built cleanly, and it works as expected following the [instructions](https://github.com/ros2/ros1_bridge).

codebot, 2016-12-13:
Linux is has one warning about defining the operator `delete[]` in `test_tlsf.cpp` in `realtime_support`. I'm working on that now.

OSX is clean: http://ci.ros2.org/job/ci_osx/1575/

Windows had one failure but seems like an unrelated flaky test; I'm running it again now.

codebot, 2016-12-13:
Clean CI:

linux: http://ci.ros2.org/job/ci_linux/2055/
os x: http://ci.ros2.org/job/ci_osx/1588/
windows: http://ci.ros2.org/job/ci_windows/2071/

Merged on all repos now.

closed 2016-12-13 by codebot

Sarcasm, 2016-12-17:
Just a thought, any reason you prefer to specify`-std=c++14` flag directly instead of relying on the CMake features?

For example, one of the solution to enable C++14 could be:

```cmake
set(CMAKE_CXX_STANDARD 14)
set(CMAKE_CXX_STANDARD_REQUIRED ON)
set(CMAKE_CXX_EXTENSIONS OFF)  # optional
```

There are other methods, more fine grained but more work, I think this article gives a good overview:

- https://cmake.org/cmake/help/v3.1/manual/cmake-compile-features.7.html
- https://crascit.com/2015/03/28/enabling-cxx11-in-cmake/

wjwwood, 2016-12-18:
@Sarcasm that's a good point, I was actually just looking at those for a different reason, but I didn't want to hold up the C++14 branches to change to that since it's untested in our code so far. When we first started adding the `-std=c++11` option we were still using CMake 2.8.something and it didn't have this.

We could switch to it now.