ros2/ros2 #298: API docs don't link correctly between packages

gerkey, 2016-12-20:
The docs at http://docs.ros2.org/beta1/ contain some invalid links. It looks like when we're linking to a file in another package, the link ends up with twice as many auto-generated underscores as it should have.

gerkey, 2016-12-20:
E.g., this page: http://docs.ros2.org/beta1/api/rcl/error__handling_8h.html links to this page: http://docs.ros2.org/beta1/api/rmw/error____handling__8h.html

wjwwood, 2016-12-20:
Converting http://docs.ros2.org/beta1/api/rmw/error____handling__8h.html => http://docs.ros2.org/beta1/api/rmw/error__handling_8h.html works.

mikaelarguedas, 2018-03-08:
It looks llike this has been fixed since as I couldn't find any broken links in the ardent docs.

closed 2018-03-08 by mikaelarguedas