ros2/ros2 #303: Circular dependency when building with rmw_opensplice_cpp

dhood, 2017-01-02:
Builds are not working with opensplice at the moment, e.g. [image]

```
00:00:45.872 ==> "/home/rosbuild/ci_scripts/venv/bin/python" -u "src/ament/ament_tools/scripts/ament.py" build "src" --build-tests --build-space "build" --install-space "install" --isolated --parallel
00:00:46.301 ament build: Circular dependency within the following packages: rmw_implementation, rmw_opensplice_cpp
```

Without looking into it too much (since support for opensplice is currently paused) it seems that this change was introduced some time between the beta 1 release ([image]) and 3 Jan 2017.

dirk-thomas, 2017-01-03:
The dependency direction between typesupport packages and rmw implementations has been changed in ros2/rosidl_typesupport#7. Since support for OpenSplice is paused it was not updated.

I would suggest to close this ticket since there are many things with both paused rmw implementations which don't work at the moment which are all not tracked by tickets (and I don't think it makes sense to track them while support is paused).

dhood, 2017-01-03:
My reasoning is that this ticket is for a) visibility to users and b) for developers, keeping track of when issues were introduced for if/when support is resumed.

I agree that there is no need to ticket all of the individual issues. I see this is a "milestone" of sorts. I will make the name more general as with https://github.com/ros2/rmw_connext/issues/204

BrannonKing, 2017-03-28:
I don't see an issue with circular dependencies at the moment, but I have these issues: every time I compile ROS2 it recompiles all the OpenSplice message implementations even when nothing has changed, both C and C++. It's painful because it takes an hour. 

In addition, I can't get rclpy to work with it at all; it builds the agnostic target but later complains like this: RuntimeError: Failed to create publisher: type support handle implementation 'rosidl_typesupport_c' (0x7f86492bf850) does not match valid type support ('rosidl_typesupport_opensplice_cpp' (0x7f864b8da650))

wjwwood, 2017-03-28:
@BrannonKing we're currently not supporting OpenSplice, but if you are interested in using it, you can feel free to open issues and pull requests on the actual `rmw_opensplice` repository:

https://github.com/ros2/rmw_opensplice

In fact, @dhood shouldn't this be on that repository? Or did you have some other reason for putting it on ros2/ros2? I know you mentioned that you felt it was a milestone, but it's not clear to me what that means in terms of being on ros2/ros2 versus ros2/rmw_opensplice.

dhood, 2017-03-29:
it's here for visibility to users who don't know to look on other repos for open issues

mikaelarguedas, 2017-08-08:
given that [last build](http://ci.ros2.org/job/ci_linux/2922/#showFailuresLink) built without warning and "only" 34 tests failed I'm going to close this.

closed 2017-08-08 by mikaelarguedas