ros2/ros2 #305: compile with stricter compiler options where it's not too cumbersome

mikaelarguedas, 2017-01-05:
follow up of https://github.com/ros2/rclcpp/pull/253
I created pedantic branch for most ros2 specific repos. This branch simply passes `-Wpedantic` to GCC and CLANG

The goal of the issue is to keep track of which packages don't comply with stricter rules and the reason why. This is also a place to gather feedback on which ones are worth changing to comply with stricter compiler rules.

Packages not patched (yet ?):
- **class_loader** and every package depending on it: 
  - trying to diverge as less as possible from master to facilitate backports later on
- every **message package**:
  - pedantic doesn't like casting pointers <=> function pointers. This is used heavily in python generated messages to convert data structures
- **rmw_opensplice_cpp**:
  - suport paused and currently broken
- **rcl**: 
  - the memory tools redefine a bunch of memory functions (malloc, realloc etc) but without the proper flags and attributes so build fails with pedantic. It may be cumbersome to fix if we need to add compiler/platform specific logic to support this flag
- **tlsf_cpp**:
  - uses VLAs, should be fixable but haven't find an elegant way to do it yet

Except from those packages, CI is passing without warning on gcc and clang.
Linux: [image]
OSX: [image]

I also gave a try to also pass /W4 to MSVC
as expected MSVC is spitting a lot more warnings on about every package. Some of them are legit and will be addressed. But I think that a good amount could/should be ignored.
Windows: [image]

mikaelarguedas, 2017-02-10:
Here is a a first iteration for pedantic (Linux, OSX), I put only commit whent we just have to add the flag and opened PR where there is code change.
- cli_tools: https://github.com/ros2/cli_tools/pull/4
- demos: https://github.com/ros2/demos/pull/113
- examples: https://github.com/ros2/examples/commit/13299171beb6cec8d3b5c29275f471ea29d8f01f
- rclcpp: https://github.com/ros2/rclcpp/pull/306
- rmw: https://github.com/ros2/rmw/pull/91
- rmw_connext: https://github.com/ros2/rmw_connext/pull/218
- rmw_fastrtps: https://github.com/ros2/rmw_fastrtps/pull/89
- rmw_implementation: https://github.com/ros2/rmw_implementation/commit/12a546679c2ce762c81c0c9247bc9ca4ef873c6f
- rosidl: https://github.com/ros2/rosidl/pull/207
- rosidl_typesupport: https://github.com/ros2/rosidl_typesupport/commit/832f2401467e2e011d26dbf0a246eb0215f08e31
- system_tests: https://github.com/ros2/system_tests/pull/189
- tutorials: https://github.com/ros2/tutorials/commit/86d1f790428582bd55bf4f2013cdb5a8b0846667

wjwwood, 2017-02-28:
I only had comments on one of the linked set of changes, lgtm.

mikaelarguedas, 2017-02-28:
reran CI for all these branch:
- Linux:
  - [image]
- macOS:
  - [image] (expected because of python3.6 issue)
- Windows:
  - [image]

mikaelarguedas, 2017-03-02:
Iterating on this (updating the first comment of this issue as things moves forward):
**message packages** can use this flag for c++, the blocker is still that the Cpython code uses object pointer to function pointer conversion for all the type conversion functions:
geometry2: https://github.com/ros/geometry2/commit/72b7072fa80b78909e1404b459ea7787cc1fb4fb
common_interfaces: https://github.com/ros2/common_interfaces/commit/4a5c9e3801eb20936508260be6ea52f182bb8d24
rcl_interfaces: https://github.com/ros2/rcl_interfaces/commit/dd782481e4cda0538822c0a42f4a23bf09e21bdb

One demo was not using extra or pedantic:
demos: https://github.com/ros2/demos/commit/af0699dddd60b6b7410c6200183f0a0d8ef9eff8

rmw_connext_dynamic_cpp: https://github.com/ros2/rmw_connext/pull/219

rmw_opensplice (now that it compiles again): https://github.com/ros2/rmw_opensplice/pull/163

CI (without tests only to identify warnings):
- Linux:
  - [image]
- macOS:
  - [image]

mikaelarguedas, 2017-03-24:
CI with rebased branches and addressed review comments:
- Linux:
  - [image]
- macOS:
  - [image]

mikaelarguedas, 2018-03-08:
class_loader now complies as well: https://github.com/ros/class_loader/pull/80

mikaelarguedas, 2018-09-20:
Message packages now compile with pedantic:
https://github.com/ros-planning/navigation_msgs/commit/7a7d12c86a3933fae142263b4c337948e606c749
https://github.com/ros2/demos/commit/0ca06c9946157a1a592f038b1a876a2412b0241d
https://github.com/ros2/example_interfaces/commit/aebff129144c968ba5b49ac8cd15ed235cadd3ca
https://github.com/ros2/geometry2/commit/80ee9d678babfca30f27e700a7825509a5d866bb
https://github.com/ros2/rcl_interfaces/commit/5b561bfd24f709f678324299d57eb14257c88f85
https://github.com/ros2/rclcpp/commit/f6c2f5ba0d12968e38d340c1857a766093e42ee1
https://github.com/ros2/rcutils/commit/1bb9cc611fe63ef83d6e4335c3f4800328492272
https://github.com/ros2/system_tests/commit/d54981b475c7f3115a0c1e62fabc25c03d1b5c15
https://github.com/ros2/common_interfaces/pull/59

CI at: https://github.com/ros2/common_interfaces/pull/59#issue-216485874

clalancette, 2021-01-07:
Between the work that Mikael did here, and the more recent work that Audrow did, we've enabled a lot of warnings on a lot of packages.  While we can go further, that is always the case, so I'm going to consider this "done".  If there are additional flags we should enable on specific packages, we'll open up separate issues for each of them for easier tracking.

closed 2021-01-07 by clalancette