ros2/ros2 #306: Control shared/static linking via BUILD_SHARED_LIBS and ament_cmake_ros

esteve, 2017-01-12:
Just an issue to aggregate all the connected PRs so that it shows up on the waffle board

dirk-thomas, 2017-04-07:
The following CI jobs cover:

* ament/gtest_vendor#4
* ament/gmock_vendor#3
* ros2/rmw#98
* ros2/rcl#115

---

* [image]
* [image]
* [image]

dirk-thomas, 2017-04-12:
@esteve Waiting for updates / responses on the following PRs:

* ros2/rosidl#183
* ros2/rosidl_typesupport#10
* ros2/rmw_fastrtps#87

dirk-thomas, 2017-04-14:
* [image]
* [image]
* [image]

esteve, 2017-04-14:
@dirk-thomas I assume you didn't pass `BUILD_SHARED_LIBS` to the CI build (i.e. the build is using the default `SHARED` value), right? If that's the case, it seems the code in https://github.com/ros2/rosidl_typesupport/pull/10 doesn't read the `BUILD_SHARED_LIBS` variable as expected. Should `ament_cmake_ros` be added as an export dependency and added to https://github.com/ros2/rosidl_typesupport/blob/master/rosidl_typesupport_c/rosidl_typesupport_c-extras.cmake.in and https://github.com/ros2/rosidl_typesupport/blob/master/rosidl_typesupport_cpp/rosidl_typesupport_cpp-extras.cmake.in ?

dirk-thomas, 2017-04-14:
I am not sure if exporting `ament_cmake_ros` is a good idea. In that case a package might not directly depend on `ament_cmake_ros` but implicitly use it through the typesupport dependencies. I would think that users will get confused why their by default statically linked libraries are suddenly linked shared.

esteve, 2017-04-17:
@dirk-thomas that's a good point, I hadn't thought about that. I've pushed https://github.com/ros2/rosidl_typesupport/pull/10/commits/853357e532322b64040f7b636855dcc0c4bfa3f9 with a workaround, I don't know if it'd suffice, but that logic should only take effect if the user deliberately chooses to build libraries as static by setting `BUILD_SHARED_LIBS` to `OFF` / `FALSE` explicitly.

dirk-thomas, 2017-04-20:
Please see https://github.com/ros2/rosidl_typesupport/pull/10#issuecomment-295814568 for the latest comment on this.

dirk-thomas, 2017-06-07:
Any news on this?

esteve, 2017-06-08:
@dirk-thomas see https://github.com/ros2/rosidl/pull/183#issuecomment-307049531

mikaelarguedas, 2017-07-26:
@esteve I'm going to put this in backlog due to the low activity and the fact that we are not planning on allocating time to this in the next cycle. We will still happily review and test changes to the referenced PRs

esteve, 2017-07-29:
@mikaelarguedas the last missing piece of this is https://github.com/ros2/rosidl/issues/183, I proposed a solution but I don't know if that's the correct one (https://github.com/ros2/rosidl/pull/183#issuecomment-307049531). I'll gladly work on it, but @dirk-thomas seemed to have his doubts about the proposed solution in https://github.com/ros2/rosidl/pull/183#issuecomment-307163340

I can go ahead and implemented what I proposed (adding `ament_cmake_ros` to message packages) if it has a chance of being considered for merging. But if there's any other proposal, I can implement that either.

dirk-thomas, 2018-03-01:
I am closing this in favor of #468 which aggregates the PRs which are currently being worked on.

closed 2018-03-01 by dirk-thomas