ros2/ros2 #307: Mixed types between Pub/Sub and Services

Karsten1987, 2017-01-21:
when running `rostopic_list`, which essentially only prints available topics and their types, I see this inconsistency between pub/sub and services. As I don't know exactly where to assign this issue at, I'll do it here:
```
Name: chatter Type: std_msgs/String
Name: lc_talker__change_stateReply Type: lifecycle_msgs::srv::dds_::ChangeState_Response_
Name: lc_talker__change_stateRequest Type: lifecycle_msgs::srv::dds_::ChangeState_Request_
Name: lc_talker__get_available_statesReply Type: lifecycle_msgs::srv::dds_::GetAvailableStates_Response_
Name: lc_talker__get_available_statesRequest Type: lifecycle_msgs::srv::dds_::GetAvailableStates_Request_
Name: lc_talker__get_available_transitionsReply Type: lifecycle_msgs::srv::dds_::GetAvailableTransitions_Response_
Name: lc_talker__get_available_transitionsRequest Type: lifecycle_msgs::srv::dds_::GetAvailableTransitions_Request_
Name: lc_talker__get_stateReply Type: lifecycle_msgs::srv::dds_::GetState_Response_
Name: lc_talker__get_stateRequest Type: lifecycle_msgs::srv::dds_::GetState_Request_
Name: lc_talker__transition_event Type: lifecycle_msgs/TransitionEvent
Name: parameter_events Type: rcl_interfaces/ParameterEvent
Number of currently available topics: 11
```

mikaelarguedas, 2017-02-03:
If looks like this come from a name conversion that happens only for messages topics for some reason, see 
https://github.com/ros2/rmw_opensplice/blob/master/rmw_opensplice_cpp/src/rmw_topic_names_and_types.cpp#L102-L108
https://github.com/ros2/rmw_connext/blob/master/rmw_connext_shared_cpp/src/shared_functions.cpp#L836-L843
https://github.com/ros2/rmw_fastrtps/blob/master/rmw_fastrtps_cpp/src/functions.cpp#L2298-L2306

The question is how do we expect service topics types to be represented in the ROS world?
should `lifecycle_msgs::srv::dds_::ChangeState_Response_` be `lifecycle_msgs/ChangeState_Response` ?

mikaelarguedas, 2017-08-08:
@Karsten1987 can you confirm that this is fixed since https://github.com/ros2/rmw_fastrtps/pull/119 (and connext equivalent)?

Karsten1987, 2017-08-08:
positive!

closed 2017-08-08 by Karsten1987