ros2/ros2 #308: CMake build type wiki change

wjwwood, 2017-02-02:
@dclanmaster I reverted your change to the wiki because we believe it shouldn't be necessary. According to the CMake documentation it should be `Debug`:

https://cmake.org/cmake/help/v3.5/variable/CMAKE_BUILD_TYPE.html

It is also our team's experience that it should be capitalized, e.g. `Debug` or `RelWithDebInfo`. Perhaps there is some other issue at play here.

I'm going to close this (opening it was the only way I could think to address the feedback to you), but if you're still seeing the issue and would like help resolving it or disagree with us, please comment here and I'll reopen it.

wjwwood, 2017-02-02:
Oh and here is the revert:

https://github.com/ros2/ros2/wiki/Ament-Tutorial/_compare/4ef4c9a1d9ba7ddbebc7b527e57a42ffd9a30941...b438c05b20b23760d770567bcc759e6f1d7d3137

And the original change:

https://github.com/ros2/ros2/wiki/Ament-Tutorial/_compare/583a214e745c1f886d46dc594b5df66f4f5f701b...4ef4c9a1d9ba7ddbebc7b527e57a42ffd9a30941

closed 2017-02-02 by wjwwood

mikaelarguedas, 2017-02-02:
@dclanmaster if you provide a `CMAKE_BUILD_TYPE` that is not `Debug`, `Release`, or `RelWithDebInfo` the variable will be ignored and cmake will fall back to the default behavior.
So if you experienced errors while building with `Debug` but not with `debug` that could be because you are missing some debug symbols. We would be glad to learn more about what the exact error was and update our documentation accordingly

dclanmaster, 2017-02-02:
Sorry for rude changing! didn't realize to put issue here. Thank you for feedback, lesson learned. Reason of fail to build is "python_d.exe" missing from chocolatey python36. Tried reinstall from chocolatey. Tried installing from the new tarball  "python36". And it works now

mikaelarguedas, 2017-02-02:
There are installation instructions for installing python debug libraries at the bottom of the installation page in the ["extra stuff for debug mode" section"](https://github.com/ros2/ros2/wiki/Windows-Development-Setup#extra-stuff-for-debug-mode), I guess they lack visibility. Do you have suggestion on how we can make them more visible to users ?
Glad it works, thanks for the feedback