ros2/ros2 #309: console_bridge is not installed properly

mkhansen-intel, 2017-02-03:
I installed ros2 from binary following the installation instructions. Now I am trying to set up and use ament to build the demos package so I can do some hacking of the demos. I followed the directions to set up ament by installing all the dependencies, and I cloned the ros2/demos repo into my ros2_ws/src path. Now I am trying to build the demos by running 'ament build'. I get the following error:

CMake Error at /home/mkhansen/ros2_install/ros2-linux/lib/console_bridge/cmake/console_bridge-config.cmake:15 (message):
  Library 'console_bridge' in package console_bridge is not installed
  properly

Now, I fully admit I may have done something wrong. The tutorials aren't clear on exactly what steps are needed and exactly how to run ament, so I followed these steps:
https://github.com/ros2/ros2/wiki/Linux-Development-Setup#how-to-setup-the-development-environment

However, I did NOT 'wget https://raw.githubusercontent.com/ros2/ros2/release-latest/ros2.repos' since I already installed from binary. I just want to hack the demos, so I cloned those instead.

Can someone tell me why console_bridge is not installed properly?

wjwwood, 2017-02-03:
I'm working on reproducing this now, to see what the exact issue is, but what I would caution you about with your current approach is that you're mixing the contents of beta 1 binary with the source from master of some repositories. This *may* be ok for the demos, but in general it is dangerous since we've changed several things since the beta 1 and you might be some inconsistency by mixing and matching different versions of software like this. For example, there are new changes on the demos repository already which may or may not work with the binaries from `release-beta1`:

https://github.com/ros2/demos/compare/release-beta1...master

You could clone the demos repository and change to the `release-beta1` tag before starting your hacking, but that might make it harder to contribute a pull request afterwards if that is your goal.

wjwwood, 2017-02-03:
I can reproduce this issue, but I'm still looking at why it's happening. This is what I did:

- Download the beta1 release for linux from: https://github.com/ros2/ros2/releases/tag/release-beta1
- Extracted it to `~/testing/ros2-linux`
- Sourced `~/testing/ros2-linux/setup.zsh` (I'm using `zsh`, but that shouldn't matter)
- `mkdir -p ~/testing/demos_ws/src`
- `cd ~/testing/demos_ws/src`
- `git clone https://github.com/ros2/demos.git`
- `cd ..`
- `ament build`

This is the error I get:

```
% ament build
# Topological order
 - composition
 - demo_nodes_cpp
 - demo_nodes_py
 - image_tools
 - intra_process_demo
 - lifecycle
 - pendulum_msgs
 - pendulum_control
 - topic_monitor

Process package 'composition' with context:
--------------------------------------------------------------------------------
 source_space => /home/william/testing/demos_ws/src/demos/composition
  build_space => /home/william/testing/demos_ws/build/composition
install_space => /home/william/testing/demos_ws/install
   make_flags => -j4, -l4
  build_tests => False
--------------------------------------------------------------------------------
+++ Building 'composition'
Running cmake because arguments have changed.
==> '. /home/william/testing/demos_ws/build/composition/cmake__build.sh && /usr/bin/cmake /home/william/testing/demos_ws/src/demos/composition -DBUILD_TESTING=0 -DCMAKE_INSTALL_PREFIX=/home/william/testing/demos_ws/install' in '/home/william/testing/demos_ws/build/composition'
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found ament_cmake: 0.0.0 (/home/william/testing/ros2-linux/share/ament_cmake/cmake)
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.5.2", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Found ament_index_cpp: 0.0.0 (/home/william/testing/ros2-linux/share/ament_index_cpp/cmake)
-- Found class_loader: 1.0.0 (/home/william/testing/ros2-linux/share/class_loader/cmake)
CMake Error at /home/william/testing/ros2-linux/lib/console_bridge/cmake/console_bridge-config.cmake:15 (message):
  Library 'console_bridge' in package console_bridge is not installed
  properly
Call Stack (most recent call first):
  /home/william/testing/ros2-linux/share/class_loader/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /home/william/testing/ros2-linux/share/class_loader/cmake/class_loaderConfig.cmake:30 (include)
  CMakeLists.txt:11 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/william/testing/demos_ws/build/composition/CMakeFiles/CMakeOutput.log".

<== Command '. /home/william/testing/demos_ws/build/composition/cmake__build.sh && /usr/bin/cmake /home/william/testing/demos_ws/src/demos/composition -DBUILD_TESTING=0 -DCMAKE_INSTALL_PREFIX=/home/william/testing/demos_ws/install' failed in '/home/william/testing/demos_ws/build/composition' with exit code '1'
<== Command '. /home/william/testing/demos_ws/build/composition/cmake__build.sh && /usr/bin/cmake /home/william/testing/demos_ws/src/demos/composition -DBUILD_TESTING=0 -DCMAKE_INSTALL_PREFIX=/home/william/testing/demos_ws/install' failed in '/home/william/testing/demos_ws/build/composition' with exit code '1'
```

I'll take a moment to see if the issue is something that affects beta1 or master, but I'll reiterate that this workflow is unsupported in general.

wjwwood, 2017-02-03:
I get the same error if I use the `release-beta1` tag for the demos repository.

wjwwood, 2017-02-03:
I do not see the error if I build ros2 from source and then build the demos repository in another workspace on top.

This means to me that either the bug exists in beta1 but has since been fixed or there is a problem with our packaging process. Both are possible, but if there is a bug like this in beta1, we might choose not to try and fix it since it's fixed in master. If it's a bug in our packing process then we'll probably look into it sooner.

I'll try a newer binary from our packaging job (something like a nightly) to see if I can reproduce it with that.

---

This is what I did to set up the "from source" workspace:

- `mkdir -p ~/testing/default/ros2_ws/src`
- `cd ~/testing/default/ros2_ws`
- `wget https://raw.githubusercontent.com/ros2/ros2/master/ros2.repos`
- `vcs import ./src < ros2.repos`
- `./src/ament/ament_tools/scripts/ament.py build`
- `source ./install/setup.zsh`
- `mkdir -p ~/testing/default/demo_ws/src`
- `cd ~/testing/default/demo_ws/src`
- `git clone https://github.com/ros2/demos`
- `cd ..`
- `ament build`

wjwwood, 2017-02-03:
I can reproduce it with one of our nightlies, so this must be some sort of a bug in our packaging process. I'll make a new issue about this with instructions on how to reproduce it.

In the meantime, I recommend you build it from source. It shouldn't take very long (about 10 minutes on my machine). Thanks for the bug report.

wjwwood, 2017-02-03:
I'm going to close this in favor of https://github.com/ros2/ros2/issues/310.

closed 2017-02-03 by wjwwood

mkhansen-intel, 2017-02-03:
@wjwwood - thanks for looking into this so quickly! I think that the usage I am attempting should be supported, which is a mix of debian and source. Granted, you're 100% correct that should have checked out the beta1 tag of the demos repo, but in that case, it should work, right? I just want to clarify, as this is something I do all the time with ROS and catkin. I install ROS from debian then pull source for packages I want to change.

wjwwood, 2017-02-03:
If you use the binaries and the `release-beta1` tag for anything you're cloning then yes it will work. The other case, i.e. use the binaries but then build other packages from source on top, is not supportable in my opinion. You can do it sometimes, but you always run the risk of skewed changes not lining up. You can do this in ROS 1 only because ROS 1 changes so slowly, but it is still a risk there as well.

mkhansen-intel, 2017-02-06:
I think we're agreeing. I should be able to mix the two (binary and source) as long as I build with the proper release tag (release-beta1 in this case. Thanks again!