ros2/ros2 #310: console_bridge library not found when chaining workspace using binaries

wjwwood, 2017-02-03:
When using the binaries we produce for Linux and when building a workspace chained off the workspace provided by those binaries, the `components` package (and perhaps others) fail to find `console_bridge`'s library generating this CMake error:

```
CMake Error at /home/william/testing/nightly/ros2-linux/lib/console_bridge/cmake/console_bridge-config.cmake:15 (message):
  Library 'console_bridge' in package console_bridge is not installed
  properly
Call Stack (most recent call first):
  /home/william/testing/nightly/ros2-linux/share/class_loader/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /home/william/testing/nightly/ros2-linux/share/class_loader/cmake/class_loaderConfig.cmake:30 (include)
  CMakeLists.txt:11 (find_package)

```

I used the most recent binary package from our packaging job ([job-#569](http://ci.ros2.org/view/packaging/job/packaging_linux/569/)) on the build farm to reproduce this, but it was originally reported in https://github.com/ros2/ros2/issues/309 with the binaries for `release-beta1`. These are the steps to reproduce the issue:

- `mkdir -p /tmp/nightly/demos_ws/src`
- `cd /tmp/nightly`
- `wget http://ci.ros2.org/view/packaging/job/packaging_linux/lastSuccessfulBuild/artifact/ws/ros2-package-linux.tar.bz2`
- extract `ros2-package-linux.tar.bz2` to the current directory
- `source ./ros2-linux/setup.zsh`
- `cd ./demos_ws/src`
- `git clone https://github.com/ros2/demos.git`
- `cd ..`
- `ament build`

At this point it fails to build with the above error. I also tried with a different build type:

- `rm -rf build`
- `ament build --cmake-args -DCMAKE_BUILD_TYPE=Release --`

This also fails.

wjwwood, 2017-02-03:
Thanks @mkhansen-intel for the original report.

mikaelarguedas, 2017-05-24:
validated that https://github.com/ros/console_bridge/pull/45 fixes this problem. But it hasn't been merged yet

PR #383 (2017-08-07): use upstream console bridge

https://github.com/ros/console_bridge/pull/45 has been merged, we can now track upstream again.

* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

Packaging:

* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

I tested locally to build an overlay with the artifact produces but the packaging job and it works.

connects to #310 
fixes #310
closed 2017-08-07 by mikaelarguedas