ros2/ros2 #311: [windows] C:¥dev¥ros2¥local_setup.bat can not find Python36

sugarsweetrobotics, 2017-02-13:
I installed ROS2 in Windows10 x64 Pro with Chocolatey v0.10.3.
C:¥dev¥ros2¥local_setup.bat can not find Python.exe because the default folder where the Python36 is installed by choco is different from "C:¥ProgramData¥Chocolatey¥lib¥Python3¥tools¥Python.exe".

I edited line 26 of local_setup.bat like below:

-  set "ament_python_executable=C:\ProgramData\chocolatey\lib\python3\tools\python.exe"
 + set "ament_python_executable=C:\Python36\python.exe"

Then, I got the correct output of the talker/listener examples.

mikaelarguedas, 2017-02-13:
Hi @sugarsweetrobotics,
Thanks for reporting the issue.

Did you install from source or from binaries?
- If from binaries: 
It may be because our windows machines building the binaries need to be updated to use the latest version of chocolatey + python.
- If building from source:
That may be because you had an older version of chocolatey and python when you built your ROS2 workspace. Since then installed/upgraded your python through chocolatey before sourcing your `local_setup.bat`

mikaelarguedas, 2017-02-13:
Note: the path to python you can set the environment variable `AMENT_PYTHON_EXECUTABLE` to override whatever Python executable was used at build time

mikaelarguedas, 2017-08-08:
duplicate of #360

closed 2018-03-30 by dirk-thomas