ros2/ros2 #312: ament-tools not found on Windows

mfolsom3, 2017-02-17:
After following the ROS 2.0 install instructions, the examples ran fine.
However, I cannot build my own.  Anytime I run ament, I get the below error.

pkg_resource:DistributionNotFound: Then ament-tools==0.0.0 distribution was not found and is required by the appliction

When I looked in the bin directory I did not see ament-tools

dirk-thomas, 2017-02-17:
Can you please link to the installation instructions. That clarifies what steps you followed.

Please provide the exact command you ran as well as the full output. I sounds like you invoked `ament` which means that executable is available. There is no `ament-tools` executable in `bin` - it is named just `ament`.

What is the environment when you ran the command? Did you source the setup file before? Specifically what is the environment variable `PYTHONPATH` set to?

mfolsom3, 2017-02-17:
I followed this link
https://github.com/ros2/ros2/wiki/Windows-Install-Binary
Every thing worked as described.

Then I tried this tutorial (Develop your own package)
https://github.com/ros2/ros2/wiki/Ament-Tutorial

Below is the result of running the command:
c:\dev\ros2_VARRI_ws>ament build --cmake-args -DCMAKE_BUILD_TYPE=Debug
Traceback (most recent call last):
  File "C:\dev\ros2\bin\ament-script.py", line 6, in <module>
    from pkg_resources import load_entry_point
  File "C:\ProgramData\chocolatey\lib\python3\tools\lib\site-packages\pkg_resources\__init__.py", line 3017, in <module>
    @_call_aside
  File "C:\ProgramData\chocolatey\lib\python3\tools\lib\site-packages\pkg_resources\__init__.py", line 3003, in _call_aside
    f(*args, **kwargs)
  File "C:\ProgramData\chocolatey\lib\python3\tools\lib\site-packages\pkg_resources\__init__.py", line 3030, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "C:\ProgramData\chocolatey\lib\python3\tools\lib\site-packages\pkg_resources\__init__.py", line 659, in _build_master
    ws.require(__requires__)
  File "C:\ProgramData\chocolatey\lib\python3\tools\lib\site-packages\pkg_resources\__init__.py", line 967, in require
    needed = self.resolve(parse_requirements(requirements))
  File "C:\ProgramData\chocolatey\lib\python3\tools\lib\site-packages\pkg_resources\__init__.py", line 853, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'ament-tools==0.0.0' distribution was not found and is required by the application

dirk-thomas, 2017-02-17:
> Did you source the setup file before? Specifically what is the environment variable PYTHONPATH set to?

mfolsom3, 2017-02-17:
I did run setup.bat

There is no PYTHONPATH variable.  However, my python directory is in my PATH variable

dirk-thomas, 2017-02-17:
You need to call the `setup.bat` file to set the `PYTHONPATH` (or set it up manually, but there are other variables too). Without the Python code being on the `PYTHONPATH` it can't be found.

mfolsom3, 2017-02-17:
OK I got this now.  The setup.bat in the ros2 workspace everything runs fine. Just not in the overlay.
Thanks

dirk-thomas, 2017-02-17:
If the problem is resolved for you can you please close the ticket.

If you still need help (not sure what "Just not in the overlay") please elaborate with more details. I assume you have tried to source the setup file of your overlay?

closed 2017-02-17 by mfolsom3