ros2/ros2 #313: Make error with failing unit tests

jlack1987, 2017-02-17:
I have written a very simple ament cmake package that has a unit test file with two simple unit tests. When the tests pass, everything seems to go ok. When a test fails though, I see the full test output that shows one test passed and one test fails and at the end there is a make error. 

I will also add that, and i'm sure the console output will give this away, I am running this all in a docker. The full console output is pasted below,

```bash
root@899ac93c7cf2:/ament_ws# ament test
# Topological order
 - hello_world_ament

Process package 'hello_world_ament' with context:
--------------------------------------------------------------------------------
 source_space => /ament_ws/src/hello-world-ci
  build_space => /ament_ws/build/hello_world_ament
install_space => /ament_ws/install
   make_flags => -j4, -l4
  build_tests => True
--------------------------------------------------------------------------------
+++ Building 'hello_world_ament'
==> '. /ament_ws/build/hello_world_ament/cmake__build.sh && /usr/bin/make cmake_check_build_system' in '/ament_ws/build/hello_world_ament'
==> '. /ament_ws/build/hello_world_ament/cmake__build.sh && /usr/bin/make -j4 -l4' in '/ament_ws/build/hello_world_ament'
[ 25%] Built target gtest
[ 50%] Built target hello_world_ament
[ 75%] Built target gtest_main
[100%] Built target beep
+++ Installing 'hello_world_ament'
==> '. /ament_ws/build/hello_world_ament/cmake__install.sh && /usr/bin/make install' in '/ament_ws/build/hello_world_ament'
[ 25%] Built target hello_world_ament
[ 50%] Built target gtest_main
[ 75%] Built target gtest
[100%] Built target beep
Install the project...
-- Install configuration: ""
-- Up-to-date: /ament_ws/install/bin/hello_world_ament
-- [ament] Deploying: /ament_ws/install/local_setup.bash
-- [ament] Deploying: /ament_ws/install/local_setup.sh
-- [ament] Deploying: /ament_ws/install/local_setup.zsh
-- [ament] Deploying: /ament_ws/install/setup.bash
-- [ament] Deploying: /ament_ws/install/setup.sh
-- [ament] Deploying: /ament_ws/install/setup.zsh
-- [ament] Deploying: /ament_ws/install/_order_packages.py
+++ Testing 'hello_world_ament'
==> '. /ament_ws/build/hello_world_ament/cmake__test.sh && /usr/bin/make test ARGS=-V -D ExperimentalTest --no-compress-output' in '/ament_ws/build/hello_world_ament'
Running tests...
UpdateCTestConfiguration  from :/ament_ws/build/hello_world_ament/DartConfiguration.tcl
Parse Config file:/ament_ws/build/hello_world_ament/DartConfiguration.tcl
   Site: 899ac93c7cf2
   Build name: Linux-c++
 Add coverage exclude regular expressions.
SetCTestConfiguration:CMakeCommand:/usr/bin/cmake
UpdateCTestConfiguration  from :/ament_ws/build/hello_world_ament/DartConfiguration.tcl
Parse Config file:/ament_ws/build/hello_world_ament/DartConfiguration.tcl
Test project /ament_ws/build/hello_world_ament
Constructing a list of tests
Done constructing a list of tests
Checking test dependency graph...
Checking test dependency graph end
test 1
    Start 1: beep

1: Test command: /usr/bin/python3 "-u" "/opt/hmi/ament/share/ament_cmake_test/cmake/run_test.py" "/ament_ws/build/hello_world_ament/test_results/hello_world_ament/beep.gtest.xml" "--output-file" "/ament_ws/build/hello_world_ament/ament_cmake_gtest/beep.txt" "--command" "/ament_ws/build/hello_world_ament/beep" "--gtest_output=xml:/ament_ws/build/hello_world_ament/test_results/hello_world_ament/beep.gtest.xml"
1: Test timeout computed to be: 60
1: -- run_test.py: invoking following command in '/ament_ws/src/hello-world-ci':
1:  - /ament_ws/build/hello_world_ament/beep --gtest_output=xml:/ament_ws/build/hello_world_ament/test_results/hello_world_ament/beep.gtest.xml
1: Running main() from gtest_main.cc
1: [==========] Running 2 tests from 2 test cases.
1: [----------] Global test environment set-up.
1: [----------] 1 test from TestSimple
1: [ RUN      ] TestSimple.failOnPurpose
1: /ament_ws/src/hello-world-ci/test/simple.cpp:5: Failure
1: Value of: true
1:   Actual: true
1: Expected: false
1: [  FAILED  ] TestSimple.failOnPurpose (0 ms)
1: [----------] 1 test from TestSimple (0 ms total)
1: 
1: [----------] 1 test from TestSimple2
1: [ RUN      ] TestSimple2.passingTest
1: [       OK ] TestSimple2.passingTest (0 ms)
1: [----------] 1 test from TestSimple2 (0 ms total)
1: 
1: [----------] Global test environment tear-down
1: [==========] 2 tests from 2 test cases ran. (0 ms total)
1: [  PASSED  ] 1 test.
1: [  FAILED  ] 1 test, listed below:
1: [  FAILED  ] TestSimple.failOnPurpose
1: 
1:  1 FAILED TEST
1: -- run_test.py: return code 1
1: -- run_test.py: verify result file '/ament_ws/build/hello_world_ament/test_results/hello_world_ament/beep.gtest.xml'
1/1 Test #1: beep .............................***Failed    0.13 sec

0% tests passed, 1 tests failed out of 1

Label Time Summary:
gtest    =   0.13 sec (1 test)

Total Test time (real) =   0.13 sec

The following tests FAILED:
	  1 - beep (Failed)
Errors while running CTest
Makefile:127: recipe for target 'test' failed
make: *** [test] Error 8

<== Command '. /ament_ws/build/hello_world_ament/cmake__test.sh && /usr/bin/make test 'ARGS=-V -D ExperimentalTest --no-compress-output'' failed in '/ament_ws/build/hello_world_ament' with exit code '2'
```
I have attached a zip file containing my ament package for anybody who wants to try to build it themselves.

[hello-world-ci.zip](https://github.com/ros2/ros2/files/784015/hello-world-ci.zip)

jlack1987, 2017-02-17:
I can confirm that in my host system outside of a docker I get the same thing.  To check if it's a thing with my specific package, I ran the tests on the [c_utilities](https://github.com/ros2/c_utilities) package and I get the exact same error.

wjwwood, 2017-02-22:
So, I think this is expected behavior. What would you expect to happen instead?

jlack1987, 2017-02-23:
Really? I just assumed there was something wrong when I saw,
```
Errors while running CTest
Makefile:127: recipe for target 'test' failed
make: *** [test] Error 8
``` 
for a unit test failure. Haven't been working in `ament` very long, but with `catkin` I don't feel like there would be any `make` errors or anything that would give me a line number in a make file as an error.

jlack1987, 2017-02-23:
Below is some quick output from a failing catkin test that, at least when looking at it, doesn't make me think anything has gone wrong with the build system. No `make` errors or anything like that.
```
[  PASSED  ] 18 tests.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] blahTest.blah

 1 FAILED TEST
-- run_tests.py: verify result "/home/user/git/proj/build/test_results/proj/gtest-TestTarget.xml"
[100%] Built target run_tests_proj_gtest_TestTarget
```

wjwwood, 2017-02-23:
Well comparing it to catkin you'd need to make sure you're using the `make test` target and not the `make run_tests` target. In catkin, the former uses ctest and the latter is a custom test runner that caktin had which we did not keep in ament AFAIK.

A better comparison would be to compare the behavior of a failing gtest being run with CTest (a la `make test`) in a vanilla CMake project.

I dunno, @dirk-thomas do you have any input on this? I do think we control this, it's basically the return code of `run_tests.py`.

jlack1987, 2017-02-27:
That's fair, admittedly I only have a few days experience with Ament so far. I just saw the output and after working with catkin for so long it just triggered a "That doesn't look right" reaction. If this is the intended behavior, feel free to close this out, and thanks so much for looking into it

dirk-thomas, 2017-02-27:
I agree that this is standard behavior of `ctest` / `make test`. `catkin_make test` as well as `ament test` should behave the same way since both tools are just calling the previous commands.

closed 2017-02-27 by dirk-thomas