ros2/ros2 #314: Building 'rosidl_generator_py' error

kintzhao, 2017-02-26:
when i run the commad :
 src/ament/ament_tools/scripts/ament.py build --build-tests --symlink-install

Process package 'rosidl_generator_py' with context:
--------------------------------------------------------------------------------
 source_space => /home/yhzhao/ros2/ros2_ws/src/ros2/rosidl/rosidl_generator_py
  build_space => /home/yhzhao/ros2/ros2_ws/build/rosidl_generator_py
install_space => /home/yhzhao/ros2/ros2_ws/install
   make_flags => -j1, -l1
  build_tests => True
--------------------------------------------------------------------------------
+++ Building 'rosidl_generator_py'
==> '. /home/yhzhao/ros2/ros2_ws/build/rosidl_generator_py/cmake__build.sh && /usr/bin/make cmake_check_build_system' in '/home/yhzhao/ros2/ros2_ws/build/rosidl_generator_py'
==> '. /home/yhzhao/ros2/ros2_ws/build/rosidl_generator_py/cmake__build.sh && /usr/bin/make -j1 -l1' in '/home/yhzhao/ros2/ros2_ws/build/rosidl_generator_py'
[  2%] Generating Python code for ROS interfaces
[  2%] Built target rosidl_generator_py__py
[ 18%] Built target rosidl_generator_py__rosidl_generator_c
[ 34%] Built target rosidl_generator_py__rosidl_typesupport_introspection_c
[ 36%] Built target rosidl_generator_py__cpp
[ 52%] Built target rosidl_generator_py__rosidl_typesupport_introspection_cpp
[ 68%] Built target rosidl_generator_py__rosidl_typesupport_c
[ 68%] Built target rosidl_generator_py
Scanning dependencies of target rosidl_generator_py__rosidl_typesupport_introspection_c__pyext
make[2]: *** No rule to make target 'rosidl_generator_py/rosidl_generator_py/_rosidl_generator_py_s.ep.rosidl_typesupport_introspection_c.c', needed by 'CMakeFiles/rosidl_generator_py__rosidl_typesupport_introspection_c__pyext.dir/rosidl_generator_py/rosidl_generator_py/_rosidl_generator_py_s.ep.rosidl_typesupport_introspection_c.c.o'.  Stop.
CMakeFiles/Makefile2:71: recipe for target 'CMakeFiles/rosidl_generator_py__rosidl_typesupport_introspection_c__pyext.dir/all' failed
make[1]: *** [CMakeFiles/rosidl_generator_py__rosidl_typesupport_introspection_c__pyext.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

<== Command '. /home/yhzhao/ros2/ros2_ws/build/rosidl_generator_py/cmake__build.sh && /usr/bin/make -j1 -l1' failed in '/home/yhzhao/ros2/ros2_ws/build/rosidl_generator_py' with exit code '2'
<== Command '. /home/yhzhao/ros2/ros2_ws/build/rosidl_generator_py/cmake__build.sh && /usr/bin/make -j1 -l1' failed in '/home/yhzhao/ros2/ros2_ws/build/rosidl_generator_py' with exit code '2'

dirk-thomas, 2017-02-26:
Please provide more information like what platform are you on, which instructions you have followed before this command, and what exact sources you are trying to compile.

kintzhao, 2017-02-27:
I run on Ubuntu 16.04 64bits, kinetic ROS. Run this command:

cd ~/ros2_ws
sudo sh -c 'echo "deb http://packages.ros.org/ros/ubuntu $(lsb_release -sc) main" > /etc/apt/sources.list.d/ros-latest.list'
sudo apt-key adv --keyserver hkp://pool.sks-keyservers.net --recv-key 0xB01FA116
sudo apt-get update
sudo apt-get install python-vcstool

sudo apt-get update
sudo apt-get install git wget
sudo apt-get install build-essential cppcheck cmake libopencv-dev libpoco-dev libpocofoundation9v5 libpocofoundation9v5-dbg python-empy python3-dev python3-empy python3-nose python3-pip python3-setuptools python3-vcstool
# dependencies for testing
sudo apt-get install clang-format pydocstyle pyflakes python3-coverage python3-mock python3-pep8 uncrustify
# dependencies for fastrtps
sudo apt-get install libboost-chrono-dev libboost-date-time-dev libboost-program-options-dev libboost-regex-dev libboost-system-dev libboost-thread-dev
sudo apt-get install libboost-all-dev libpcre3-dev zlib1g-dev  python-empy python-pkg-resources

wget https://raw.githubusercontent.com/ros2/ros2/master/ros2.repos
vcs import ~/ros2_ws/src < ros2.repos

src/ament/ament_tools/scripts/ament.py build --build-tests --symlink-install

dirk-thomas, 2017-02-27:
I have tried to reproduce you commands in a clean Docker container using Ubuntu Xenial. It successfully built the workspace and I could reproduce your problem.

Maybe the full output of your build contains any valuable information why it fails. Please remove the build and install folder, try another build and post the complete output e.g. in a gist. Please also make sure that you have the latest code (by invoking `vcs pull` in your workspace, `vcs status` will show if you have any local changes).

dirk-thomas, 2017-02-27:
I updated my previous comment: I could *not* reproduce the problem.

kintzhao, 2017-02-28:
Thanks, that is so amazing, I create a new ros2_ws to import  sources and build successfully (the status of  old sources  is newly and  wget  url is same with new sources).  Any way, i will start to learn more.

closed 2017-02-28 by kintzhao