ros2/ros2 #316: Correct way to import python modules.

sidd1, 2017-03-03:
Let's my package (In my ros2_overlay_ws) is setup like this -

    build/
    install/
    src/
         my_pkg/
            my_pkg/
                scripts/
                nodes/
          src
         setup.py
         README.md

I have trouble importing python modules from `my_pkg/scripts` into `my_pkg/nodes`. This is because the executable runs from the `build` folder. The workaround I have is to add a system path pointing to `my_pkg` in my node so it can find all python modules within `my_pkg`. 

I wonder if there's a more elegant way to do this.

dirk-thomas, 2017-03-03:
In order to use your Python module during the build process you need to take care that it is findable yourself. This can be done e.g. like this: https://github.com/ament/ament_tools/blob/5c0e2a5906973c4935357b7fbdf8a8a51b95c36b/scripts/ament.py#L33-L35 You should place you executable scripts into a separate folder instead of into the Python package folder in order to not install it with the Python code.

If you also need the scripts after the package has been installed (making it available to others) you should not install plain Python scripts but use entry points instead (see e.g. https://github.com/ament/ament_lint/blob/16cb042e5d04e6d50b60f4c5d373d356052370d9/ament_cpplint/setup.py#L30). This generates an executable on Windows. Otherwise a plain Python script is not easily runnable (e.g. by double clicking it).

wjwwood, 2017-03-03:
I always find this document useful:

http://setuptools.readthedocs.io/en/latest/setuptools.html#developer-s-guide

It talks about source code layout and `entry_points`.

sidd1, 2017-03-03:
Thanks that document is very helpful. But a follow up question - if I had a python api and I had to make calls to individual methods would I create an entry point for each method?

wjwwood, 2017-03-03:
@sidd1 usually not. You'd use the "console script" type entry point which would call a "main" function for each script you wanted. the main function for that script would be in a Python package or module and it would call other methods directly.

sidd1, 2017-03-03:
Got it, that makes sense. Thanks!

closed 2017-03-03 by sidd1