ros2/ros2 #320: where is 'vcs' command?

HuanjunKong, 2017-03-17:
I would like to build ros2 from source, so I checked [Linux-Development-Setup](https://github.com/ros2/ros2/wiki/Linux-Development-Setup).

I should use `vcs import src < ros2.repos` to clone all repos, but there's no `vcs` command in my Ubuntu. I have tried `apt-cache search vcs` and google.

Anybody can help me?

dhood, 2017-03-17:
I can remember having a similar question once :smile: 

The package is called `python3-vcstool`: it gets installed in the "Install GCC, G++ CMake, Python 3 EmPy package (custom packages which don't collide) and setuptools:" section of the install instructions you linked to. You might want to start from the top and make sure each line is executed.

HuanjunKong, 2017-03-17:
Thanks, I spent time to solve dependencies problem and forgot install `python3-vcstool`.

closed 2017-03-17 by HuanjunKong

gr0mph, 2018-10-31:
Hi,

Me too. On Colcon Tutorial, at 

> wget https://raw.githubusercontent.com/ros2/ros2/master/ros2.repos
> vcs import ~/ros2_ws/src < ros2.repos

vcs not found.
I have find this page:
https://index.ros.org/doc/ros2/Linux-Development-Setup/

Describe how to install vcs.
But need also other installation.

Best regards

mjcarroll, 2018-10-31:
As mentioned above, it is available through your system package manager by running: `sudo apt-get intsall python3-vcstool`

melnarte, 2020-11-25:
E: Unable to locate package python3-vcstool

nuclearsandwich, 2020-11-25:
> E: Unable to locate package python3-vcstool

@melnarte If you're having trouble with a ROS installation please ask a question on https://answers.ros.org or if you have identified a problem. Open an issue on the appropriate repository instead of commenting on a two year old issue.

Duttonide, 2020-12-11:
E: Unable to locate package python3-vcstool 
can someone give a solution to this error?

nuclearsandwich, 2020-12-11:
@Duttonide If you're having trouble with a ROS installation please ask a question on answers.ros.org or if you have identified a problem. Open an issue on the appropriate repository instead of commenting on a two year old issue.