ros2/ros2 #321: Beta 2 on the 14th of June

mikaelarguedas, 2017-03-21:
- [ ] ROS 2 native support for:
  - [ ] Cartographer
  - [ ] AMCL
  - [ ] move_base
  - [ ] astra driver
  - [ ] map_server
  - [ ] joystick_driver
  - [ ] kobuki driver
  - [ ] robot_pose_ekf or robot_localization

- [ ] Support via the ROS <--> ROS2 bridge:
  - [ ] rosbag
  - [ ] rviz

- [ ] New documentation platform deployed
- [ ] Release binary packages, at least Ubuntu
- [ ] Namespaced topics and services
- [ ] rostopic command line tool (not all verbs)
- [ ] Logging API (rosconsole)
- [ ] Parameters from a yaml file and commandline
- [ ] Command line static remapping

BrannonKing, 2017-03-21:
I have Cartographer mostly updated to ROS2 (I don't use all the features). Is there somewhere I should help or contribute some of that work for an official update? Same story on map_server.

clalancette, 2017-03-21:
Just to comment on some parts of this:

- astra driver is more-or-less done, probably needs some cleanup
- Cartographer is getting there, and will hopefully be done this week

clalancette, 2017-03-21:
@BrannonKing It looks like we have both been working on Cartographer for ROS2.  My current code is located in this fork: https://github.com/ros2/cartographer_ros.git .  Right now, I have the node starting up, but I haven't tested it further than that.  Is your code available somewhere that I can look at?  We should probably combine our work somehow.

BrannonKing, 2017-03-21:
@clalancette , this is going to be an extended conversation and some significant coordination. Can you email me at brannon.king at asirobotsdotcom?

mikaelarguedas, 2017-03-21:
thanks @BrannonKing! regarding `map_server`, there is a port available [here](https://github.com/ros2/navigation/tree/ros2/map_server), feel free to open a PR against it if there is anything you did and want to contribute

dirk-thomas, 2017-06-20:
The next milestone is to get "everything" in / merged. After that we will wait for one "clean" set of nightly builds. On the morning after that we will start the testing process by going through the excel sheet.

The remaining tasks for this next milestones are:

* [ ] FastRTPS 1.5 getting closer, all test failures fixed, make PR to address the remaining warnings (@wjwwood)
* [ ] follow up on tf2 test failures (@tfoote)
* [ ] confirm command line tools work on Windows (@dirk-thomas)
* [ ] crashes around rclpy / rmw_fastrtps_cpp (@allenh1, @mikaelarguedas)
* [ ] update dependencies on Jenkins nodes (@wjwwood)
* packaging (@nuclearsandwich)

  * [ ] get pending PRs updated and merged, remove Python dep from other packages
  * [ ] address urdfdom naming
  * [ ] kobuki and ros1_bridge dependencies
  * [ ] question about sourcing ROS 1 and ROS 2 for the demo, anybody to provide feedback

* [ ] keyboard teleop, merge as-is after feedback from @mikaelarguedas (@Kukanani)

* **anybody** [ ] update spread sheet for testing to include the new / changes stuff
* **anbody** [ ] update documentation / tutorials before the "testday" to avoid duplicate effort

Please check completed items.

nuclearsandwich, 2017-06-23:
> ros2/turtlebot2_demo#45 (however, if this is too much work for @nuclearsandwich, we can delay this)

Adding additional leaf packages at this stage isn't much trouble. Though there is an untested case in both this package and the amcl_demo package (https://github.com/ros2/turtlebot2_demo/pull/44) that is described in https://github.com/ros2/turtlebot2_demo/pull/45#issuecomment-310775502 so I'd very much like to build candidate packages of either one as soon as it's ready to know if there will be issues or not.

dirk-thomas, 2017-07-05:
:ship:

closed 2017-07-05 by dirk-thomas