ros2/ros2 #322: tf2_ros/transform_broadcaster.h does not declare all dependencies

BrannonKing, 2017-03-21:
I'm not sure which repository should hold this bug. Feel free to forward it.

The file `tf2_ros/transform_broadcaster.h` utilizes `tf2_msgs/msg/tf_message.hpp` but does not include it. I have to `#include "tf2_msgs/msg/tf_message.hpp"` before including the broadcaster for successful compilation.

I also see these two unrelated annoyance warnings. They could be fixed at the same time:
tf2/transform_datatypes.h:71:2: warning: extra ‘;’ [-Wpedantic]
tf2/buffer_core.h:416:2: warning: extra ‘;’ [-Wpedantic]

BrannonKing, 2017-03-21:
Actual compilation error:

tf2_ros/transform_broadcaster.h:72:56: error: template argument 1 is invalid
   rclcpp::publisher::Publisher<tf2_msgs::msg::TFMessage>::SharedPtr publisher_;
                                                        ^
tf2_ros/transform_broadcaster.h:72:59: error: expected ‘;’ at end of member declaration
   rclcpp::publisher::Publisher<tf2_msgs::msg::TFMessage>::SharedPtr publisher_;
                                                           ^
tf2_ros/transform_broadcaster.h:72:69: error: ‘publisher_’ does not name a type
   rclcpp::publisher::Publisher<tf2_msgs::msg::TFMessage>::SharedPtr publisher_;

sloretz, 2017-03-21:
I think the right repo for this is https://github.com/ros/geometry2

closed 2017-03-21 by sloretz

gerkey, 2017-03-21:
We fixed a bunch of problems, perhaps including this one, in ros/geometry2#208.