ros2/ros2 #323: local_setup.bat

Karsten1987, 2017-03-21:
I just compiled my workspace under windows 10, wanted to setup my environment and ran into the following error. Did anybody encounter this already?

```
c:\dev\ros2>call install_isolated\local_setup.bat
The input line is too long.
The syntax of the command is incorrect.
```

dirk-thomas, 2017-03-21:
That is a common limitation on Windows. Can you please post the line in question.

mikaelarguedas, 2017-03-21:
Unfortunately that's a known issue. I started to look into using powershell for everything. That'll allow us to overcome any limitation due to cmd. It won't solve the maximum `PATH` length though (AFAIK)

Karsten1987, 2017-03-21:
https://support.microsoft.com/en-us/help/830473/command-prompt-cmd.-exe-command-line-string-limitation

Karsten1987, 2017-03-21:
I don't have the line in question, but I am almost certain it's the `PATH` variable, which contains more than 8192 characters.

mikaelarguedas, 2017-03-21:
Yeah I think that's the issue too. Quick hack is to not build in isolation, this way you add a single folder to your PATH rather than N (where N number of packages)

Karsten1987, 2017-03-21:
So i can confirm that building in non-isolated prevents that long `PATH` variable to be set. This may be important when building the binary packages for Windows.

mikaelarguedas, 2017-09-02:
this should be fixed once we move away from batch in favor of powershell so I'm going to connect theses issues together

closed 2023-03-30 by clalancette