ros2/ros2 #326: Unable to view `std_msgs.String` msgs on Python, but able to view on C++?

sidd1, 2017-03-24:
I have a C++ node which publishes a string of characters. If I have a subscriber in C++, I'm able to read the messages. However if I have a similar publisher on python I get an assert error - 
`build/std_msgs/rosidl_generator_py/std_msgs/msg/_string_s.c:54: std_msgs_string__convert_to_py: Assertion 'pydata != NULL' failed`. 

Can someone tell me what could be wrong?

mikaelarguedas, 2017-03-24:
It looks like the c structure received cannot be casted properly by the C python extensions. Do you have the same problem with basic nodes sending std_msgs/String (e.g running the c++ `talker` and the python `listener_py`)?.
Could you send a sample of the code you use for creating and sending your String objects?

sidd1, 2017-03-24:
I am compressing a JPEG img and sending the compressed data as a str. It's something like this - 

    std_msgs::msg::Int8 compressed_img;
    uint8_t* compressed;
    // Compress data //
    compressed_str.data = (char*)compressed;
    pub.publish(compressed_str);

mikaelarguedas, 2017-03-24:
ok so that's the reason it's not failing. The `String` message assumes [ASCII characters](http://design.ros2.org/articles/interface_definition.html) and is decoded as such. So any byte having a value not in represented as ascii will fail (PyUnicodeFromString line 53 of build/std_msgs/rosidl_generator_py/std_msgs/msg/_string_s.c is failing and returning a NULL pointer).

To send series of bytes that are not ascii, please consider using a `uint8[]` type that is more suitable for representing a array of values in the range [0:255].

sidd1, 2017-03-24:
Oh I see. Is there a reason why c++ can handle this though?

mikaelarguedas, 2017-03-24:
c++ is using std::string that is more flexible. This is also the reason why c++ can send multibyte characters strings while C and Python don't (c.f. https://github.com/ros2/rosidl/issues/208)

sidd1, 2017-03-27:
Got it, thanks! I took your advice and used the `UInt8MultiArray` instead. I had  a little bit of digging to do for converting from (char*) to a vector. For future use, if anyone gets stuck, this is what I did - 
    
    uint8_t* compressed;
    std_msgs::msg::Int8 compressed_img;
    char* buffer = (char*)compressed;
    typedef std::vector<unsigned char> bufferType;
    bufferType::size_type size = strlen((const char*)buffer);
    bufferType vec(buffer, buffer + size);
    compressed_img.data = vec;

closed 2017-03-27 by sidd1

mikaelarguedas, 2017-03-27:
Glad you got it to work, thanks for posting your solution!