ros2/ros2 #327: Implement Namespacing

Karsten1987, 2017-03-28:
I open this issue here for keeping track about connecting PRs and status updates regarding namespacing.

The implementation is w.r.t. the design page: http://design.ros2.org/articles/topic_and_service_names.html

Namespacing has to be implemented in each rmw implementation for each communication method.

<>                   | connext_static | connext_dynamic | fastrtps | opensplice (optional)
--------------|----------------|------------------|---------|---------------------
publisher        |            TBD        |     TBD                  |    TBD   |         TBD                            
subscriber      |           TBD         |     TBD                  |    TBD   |        TBD
service            |           TBD         |     TBD                  |    TBD   |        TBD
client              |           TBD         |     TBD                  |    TBD   |        TBD

I further plan on writing a general function inside rmw, which concatenates the original topic name with the prefixes for the communication method. The list of prefixes can be found here: http://design.ros2.org/articles/topic_and_service_names.html#ros-specific-namespace-prefix

Karsten1987, 2017-04-07:
linux: [image]
osx: [image]
windwos: [image]

Karsten1987, 2017-05-08:
Ok, so I believe this is up for review!

dirk-thomas, 2017-05-08:
Is the new feature / functionality covered by any test atm?

Karsten1987, 2017-05-08:
I just updated the initial description of this PR and referenced the `rcl` PR with it. I had forgotten this one. 
The namespace feature set is not so much covered by this one single test, however given that all existing pub/sub/srv/client system tests are working, I believed it is tested within these tests.

wjwwood, 2017-05-08:
@Karsten1987 I fixed all of your "connects to ..." comments as they were broken.

Karsten1987, 2017-05-10:
linux: [image]
osx: [image]
windows: [image]

Karsten1987, 2017-05-11:
I had to modify "get_topic_names_and_types" inside `rmw_fastrtps` in order to make that test pass for fastrtps. While doing, I also modified `rmw_connext` to behavior equivalently. 
When running tools like `rostopic_list` the output corresponds to the fully qualified domain name:

```
rq/my/namespace/test_namespace_client_serverRequest [example_interfaces::srv::dds_::AddTwoInts_Request_]
rr/my/namespace/test_namespace_client_serverReply [example_interfaces::srv::dds_::AddTwoInts_Response_]
rt/parameter_events [rcl_interfaces/ParameterEvent]
```

I will open another issue to refactor `get_topic_names_and_types` in order to give a list of publishers, subscribers, services and clients, because I think this exceeds the scope of this PR. 

I am running CI again to check if this has further consequences for other tests.
Linux: [image]
osx: [image]
windows: [image]

One more note, as I developed locally on my machine, I've seen that fastrtps and connext indeed to talk to each other via services. So I wonder, why we deactivated tests for this.

Karsten1987, 2017-05-12:
So there is definitely some flakiness involved, but I would say this is ready w.r.t upcoming refactor of `get_topic_names_and_types`.

wjwwood, 2017-05-12:
@Karsten1987 I'm running one more "repeated nightly" job to gauge the flakey-ness. @dirk-thomas is also looking into an issue with flakey tests related to https://github.com/ros2/rcl/pull/120 which might address your pr's. My theory is that these flakey tests are just exacerbated by the use of partitions making things take a different amount of time, exposing existing race conditions.

wjwwood, 2017-05-18:
Using this job (nightly settings for these prs):

http://ci.ros2.org/job/ci_linux/2608/testReport/

And comparing it to this nightly:

http://ci.ros2.org/view/nightly/job/nightly_linux_repeated/655/testReport/

I figured out that these tests were unique to the first job, but found them in other recent nightlies:

- `projectroot.rosidl_generator_py.test_requester_replier__Primitives__rclpy__rmw_fastrtps_cpp`
  - http://ci.ros2.org/view/nightly/job/nightly_linux_repeated/649/testReport/
- `projectroot.test_composition__rmw_fastrtps_cpp`
  - (paired with `composition.test_composition__rmw_fastrtps_cpp.xunit.missing_result`)
  - http://ci.ros2.org/view/nightly/job/nightly_linux_repeated/650/testReport/
- `test_requester_replier__Primitives__rclpy__rmw_fastrtps_cpp_.test_requester_replier`
  - http://ci.ros2.org/view/nightly/job/nightly_linux_repeated/651/testReport/

So, based on that I don't see any new flakey tests.

I'll go ahead and merge this, and continue work on the final set of prs for this feature.

closed 2017-05-18 by wjwwood