ros2/ros2 #329: rclpy on OSX?

jamesriley101, 2017-04-04:
Is is possible to use rclpy on mac osX? I've been trying on 10.12.4, on which I can successfully run ros2 commands from the terminal (at least `talker` and `lisener`. But whenever I try talker_py or try to import rclpy with Python, I get 

>cannot import name 'x_rclpy_implementation_rmw_agnostic'

which I guess makes sense because that file is a .so not a python module.

BrannonKing, 2017-04-05:
rclpy doesn't work at all for me in current builds either. It seems to ignore the RMW_IMPLEMENTATION environment variable and ends up with no implementation at all.

wjwwood, 2017-04-05:
@BrannonKing does "current builds" mean out binaries off ci.ros2.org or did you build from source?

rclpy is working fine on my machine atm, built from source on macOS.

mikaelarguedas, 2017-04-05:
Up until Beta-1, rclpy was using the `RCLPY_IMPLEMENTATION` environment variable after Beta-1 (and current master) use the same `RMW_IMPLEMENTATION` variable as rclcpp. ([more details here](https://github.com/ros2/ros2/wiki/Working-with-multiple-RMW-implementations))

BrannonKing, 2017-04-05:
Mine is built from source, and I have opensplice on my machine as well as fast-rtps. I get this error attempting to use it: RuntimeError: Failed to create publisher: type support handle implementation 'rosidl_typesupport_c' (0x7f86492bf850) does not match valid type support ('rosidl_typesupport_opensplice_cpp' (0x7f864b8da650))

mikaelarguedas, 2017-04-05:
Does is work with Fast-RTPS?
`$ RMW_IMPLEMENTATION=rmw_fastrtps_cpp talker_py`
`$ RMW_IMPLEMENTATION=rmw_fastrtps_cpp listener_py`

We don't support opensplice since before the refactor of the typesupport so I can't say if this is expected behaviour or not. But it works on all our MacOS machines with connext and fastrtps AFAICT.

wjwwood, 2017-04-05:
@BrannonKing I'm not sure why that's happening. I guess it could be related to the recent typesupport changes, but I thought OpenSplice was updated for that one... Does that always happen or just when you're using `RMW_IMPLEMENTATION=rmw_opensplice_cpp`?

BrannonKing, 2017-04-05:
It does work correctly with `rmw_fastrtps_cpp`. I thought it was all broken. Sorry; I guess it's not the same error.

mikaelarguedas, 2017-04-05:
confirmed that this is broken in opensplice: opened [this ticket ](https://github.com/ros2/rmw_opensplice/issues/166) accordingly

mikaelarguedas, 2017-04-26:
closing this in favor of https://github.com/ros2/rmw_opensplice/issues/166

closed 2017-04-26 by mikaelarguedas