ros2/ros2 #330: post-ament list handling is ambiguous

BrannonKing, 2017-04-05:
After compiling the latest code, I have to modify three files in order for my codebase to compile. I have to add `cmake_policy(SET CMP0057 NEW)` to the top of these three files:

install/share/ament_cmake_core/cmake/core/list_append_unique.cmake
install/share/ament_cmake_python/cmake/ament_python_install_module.cmake
install/share/rosidl_typesupport_c/cmake/get_used_typesupports.cmake

Without that I get Python errors about the list operator being ambiguous.

dirk-thomas, 2017-04-05:
The policy `CMP0057` was added in CMake 3.3. ROS 2 targets only CMake 3.5 and higher. If you set the minimum required version of CMake in your `CMakeLists.txt` to 3.5 you will implicitly get `NEW` for all policies includes in CMake 3.5 and don't have to enable these policies explicitly.

BrannonKing, 2017-04-05:
I'm going to close this assuming that the cmake_minimum_version will be inherited throughout the build.

closed 2017-04-05 by BrannonKing

dirk-thomas, 2017-04-05:
Yes, the root `CMakeLists.txt` decides the minimum version for all included CMake files.