ros2/ros2 #333: Need for libboost-thread-dev in Linux binaries

wjwwood, 2017-04-12:
@kenny-y I reverted your change to the wiki:

https://github.com/ros2/ros2/wiki/Linux-Install-Binary/_history

We appreciate your interest in fixing the instructions, but before we commit to them, I'd like to figure out why you think it's necessary. It might be a bug in our packaging or we may have unintentionally added a dependency there that should be needed.

Can you describe which binary packages you're using and what issue lead you to add the `libboost-thread-dev` dependency?

Thanks!

kenny-y, 2017-04-12:
```
ldd ros2_install/ros2-linux/bin/talker | grep boost
	libboost_thread.so.1.58.0 => /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.58.0 (0x00007f6df56dd000)
	libboost_system.so.1.58.0 => /usr/lib/x86_64-linux-gnu/libboost_system.so.1.58.0 (0x00007f6df54d9000)
	libboost_chrono.so.1.58.0 => /usr/lib/x86_64-linux-gnu/libboost_chrono.so.1.58.0 (0x00007f6df52d1000)
```

wjwwood, 2017-04-12:
@kenny-y which binary though? An alpha or beta-1 or a nightly maybe?

kenny-y, 2017-04-12:
Beta-1

wjwwood, 2017-04-12:
Ah, this happened when we removed OpenSplice, but Fast-RTPS is still using Boost in Beta-1. In the next beta Fast-RTPS will have been de-boost-ified I think.

I'll revert my revert. Thanks for the fix.

Also, why just `libboost-thread-dev` and not all of `libboost-dev`? I imagine the thread library pulls in system and chrono automatically. If you've tested it with `libboost-thread-dev` I'll leave it as just that.

kenny-y, 2017-04-12:
I have to admit there is a bit of uncertainty... maybe we need both? The error message was:

```
talker: error while loading shared libraries: libboost_thread.so.1.58.0: cannot open shared object file: No such file or directory
```

First I tried `sudo apt install -y libboost-dev`, the same error message is still there. So I looked again and found that the error was `boost-thread`, so I `sudo apt install -y libboost-thread-dev` and everything was up and run.

wjwwood, 2017-04-12:
@kenny-y actually, I'm surprised at your `ldd` output. Where did you download that beta-1 binary, because I just tried to double check it:

```
% wget https://github.com/ros2/ros2/releases/download/release-beta1/ros2-beta1-package-linux-fastrtps.tar.bz2
[removed]

% extract ros2-beta1-package-linux-fastrtps.tar.bz2
[removed, extract is my custom shell command that just invokes tar]

% mv ros2-linux ros2-linux-beta-1

% ldd ros2-linux-beta-1/bin/talker | grep boost

% ldd ros2-linux-beta-1/bin/talker
	linux-vdso.so.1 =>  (0x00007ffd7f386000)
	librclcpp.so => not found
	librcl.so => not found
	librmw.so => not found
	libstd_msgs__rosidl_typesupport_cpp.so => not found
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f6a97d07000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f6a97af0000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f6a97727000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f6a9741e000)
	/lib64/ld-linux-x86-64.so.2 (0x00005590bac44000)

```

I get a similar result for the nightly:

```
% wget http://ci.ros2.org/view/packaging/job/packaging_linux/643/artifact/ws/ros2-package-linux-x86_64.tar.bz2

% extract ros2-package-linux-x86_64.tar.bz2
[removed, extract is my custom shell command that just invokes tar]

% mv ros2-linux ros2-linux-2017-04-11

% ldd ros2-linux-2017-04-11/bin/talker | grep boost

% ldd ros2-linux-2017-04-11/bin/talker
	linux-vdso.so.1 =>  (0x00007ffe501fd000)
	librmw.so => not found
	librcl.so => not found
	librclcpp.so => not found
	libstd_msgs__rosidl_typesupport_cpp.so => not found
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fa0a3d8a000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fa0a3b73000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fa0a37aa000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fa0a34a1000)
	/lib64/ld-linux-x86-64.so.2 (0x0000557b55bc0000)

```

Are you sure you're not using an alpha release?

wjwwood, 2017-04-12:
Oh, I need to source the `setup.*sh`, hang on.

wjwwood, 2017-04-12:
Yeah, I see it now in beta-1:

```
$ source ros2-linux-beta-1/setup.bash 

$ ldd ros2-linux-beta-1/bin/talker | grep boost
	libboost_thread.so.1.58.0 => /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.58.0 (0x00007fd508af4000)
	libboost_system.so.1.58.0 => /usr/lib/x86_64-linux-gnu/libboost_system.so.1.58.0 (0x00007fd5088f0000)
	libboost_chrono.so.1.58.0 => /usr/lib/x86_64-linux-gnu/libboost_chrono.so.1.58.0 (0x00007fd5086e8000)
```

Ok, I'm going to leave it as `libboost-thread-dev` and then add a note to reevaluate that particular dependency when we start using the master branch of Fast-RTPS again, at which point we'll hopefully drop the boost dependency again and we can make installing it an optional thing for older binary releases.

Thanks again.

closed 2017-04-12 by wjwwood

kenny-y, 2017-04-12:
It's `https://github.com/ros2/ros2/releases/download/release-beta1/ros2-beta1-package-linux-fastrtps.tar.bz2`. If we need to figure it out the requirement of boost and boost-thread. I can try it again on my clean VM.