ros2/ros2 #334: Instructions should mention Windows 10

dirk-thomas, 2017-04-12:
Since it is the only platform we actively test and support.

mikaelarguedas, 2017-04-12:
As of very recently we were still nightlies on Windows 8.1 machines as well (eatable). If we decide to stop supporting and testing for windows 8.1 (which I'm totally fine with :+1: ) that would allow us to homogenize buildfarm machines configuration and have more Windows 10 nodes.

clalancette, 2017-04-12:
I also think it would be OK to drop Windows 8.1 for now, and just concentrate on Windows 10.  It reduces our testing matrix, and if we really needed Windows 8.1 in the future for some reason, we could always resurrect it then.

mikaelarguedas, 2017-04-26:
We won't support windows 8.1 anymore. Let's remove it from all instructions

mikaelarguedas, 2017-05-10:
fixed in https://github.com/ros2/ros2/wiki/Windows-Install-Binary/_compare/cc18a39e48f540dffc325a6f9c5192df1e864a08...d0601d34f2d2eb7f0b98979a7ac55398aa142389 we'll remove windows 8.1 once we upgrade all the ci machines to windows 10

closed 2017-05-10 by mikaelarguedas

nuclearsandwich, 2017-05-10:
> we'll remove windows 8.1 once we upgrade all the ci machines to windows 10

Is there really a reason not to remove "support" for Windows 8.1 now? If it works incidentally while we have an 8.1 host up that's fine but I feel like my solution to any Windows 8.1-related issue would just be to reimage the CI machines and mark them wontfix.

mikaelarguedas, 2017-05-11:
I didn't do it given that very likely we'll keep testing it until at least the next build cop cycle. Given how "efficient" Microsoft is for getting people to upgrade to windows 10, I don't think many people are experimenting on 8.1. But the statement that we do build and test for windows 8.1 is still accurate at the time of the writing. I don't really care either way though.

nuclearsandwich, 2017-05-11:
> the statement that we do build and test for windows 8.1 is still accurate

This is the case, but the statement in the wiki includes the word "support" which unqualified carries the implication that we will make an effort to maintain that support.

> I don't really care either way though.

I only care because we know we're no longer planning to support Windows 8.1, it's unlikely that anyone will care about that, but I have at least one open issue where I don't yet know what Windows version the issue reporter is working with and I'd like to be able to point to an official Win 10 only statement if they're using an older version of Windows.

mikaelarguedas, 2017-05-11:
Fair enough, fixed [here](https://github.com/ros2/ros2/wiki/Windows-Install-Binary/_compare/d0601d34f2d2eb7f0b98979a7ac55398aa142389...9950351daad7d056aa93061b4f9d3bb5ca5763d0) and [here](https://github.com/ros2/ros2/wiki/Windows-Development-Setup/_compare/ee5a02e48451e8a24c317b88a8562aa79bc2ef85...66e18bdfa7f549815edd352cd8a7e671d450259d)
I also [drafted an issue template](https://github.com/ros2/ros2/pull/351) so that hopefully we'll have more information when future issues are open