ros2/ros2 #335: rclpy on Ubuntu

vito-dante, 2017-04-12:
ImportError: cannot import name '_rclpy_implementation_rmw_agnostic'

mikaelarguedas, 2017-04-12:
Hi @vito-dante,

Can you give more context please?
What version of the code are you using? from source or from binaries? If from source how did you build it ? 
What command did you run resulting in that error ?
Thanks!

mikaelarguedas, 2017-04-12:
Thanks for the update.
Are you using beta1 from binaries or did you compile from source?
And most importantly what commands did you run leading to this error ?

vito-dante, 2017-04-12:
my steps: [](https://github.com/ros2/ros2/wiki/Linux-Install-Binary)
```
mkdir -p ~/ros2_install
cd ~/ros2_install
tar xf ~/Downloads/ros2-package-linux-x86_64.tar.bz2
```
environment loaded
`source setup.zsh`

after, i ran  [](https://github.com/ros2/demos/blob/master/demo_nodes_py/topics/talker_py.py)
```
python talker_py.py
ImportError: cannot import name '_rclpy_implementation_rmw_agnostic'
```

mikaelarguedas, 2017-04-12:
ROS2 requires python3 (and doesn't run with python2).
Can you try to either run the  `talker_py` executable or to run `python3 talker_py.py` ? Feel free to post your results here if the problem persists

vito-dante, 2017-04-12:
my version is python 3.6

mikaelarguedas, 2017-04-12:
On Ubuntu `python` maps to Python 2.7, you can confirm that by running `python --version`. To use Python 3.6 you need to use the `python3` executable

mikaelarguedas, 2017-04-12:
I'm going to close this. if running the commands suggested in https://github.com/ros2/ros2/issues/335#issuecomment-293704849 don't solve the problem please comment here and I will reopen this

closed 2017-04-12 by mikaelarguedas