ros2/ros2 #336: remove boost from macOS machines and install instructions

wjwwood, 2017-04-12:
Now that the new version of Fast-RTPS is being used, I think we can remove instructions to install boost from the macOS machines, see:

https://github.com/ros2/ros2/wiki/OSX-Install-Binary (possibly others)

But I want to test it without boost installed and then also remove boost from the CI machines at the same time. This might also apply to Windows, and Linux too, but I was just going to work on macOS for now. If it can be removed there, I'll see what we can do to slim down the dependencies on the other OS's as well.

Related:

https://github.com/ros2/ros2/pull/332

mikaelarguedas, 2017-04-12:
Tested it only on linux in docker without installing any boost and it worked well ([Dockerfile available here](https://github.com/ros2/sros2/blob/master/sros2_docker/docker/sros2/Dockerfile)). Haven't tried the other platforms though

wjwwood, 2017-04-18:
I just finished testing it, and it can be removed. I don't think we should remove it from the jenkins slaves, however, because it might still be used in the turtlebot demos. I will remove it from the wiki instructions for OS X.

closed 2017-04-18 by wjwwood

mikaelarguedas, 2017-04-18:
@wjwwood Can you make this conditional on the version in the installation instructions? I think that the instruction link to beta1 repos file that uses a version of FastRTPS that still depends on boost

mikaelarguedas, 2017-04-18:
FYI: I updated the from source instructions to install boost for beta-1 and older https://github.com/ros2/ros2/wiki/OSX-Development-Setup/_compare/96a1f5a490621f00bad4fda8df8aac20bc0abbed...b0a6b7d7936f08ab5111a8f6f38f242ec7dcdcae