ros2/ros2 #337: In Linux Development Setup page, add missing dependencies

kenny-y, 2017-04-13:
https://github.com/ros2/ros2/wiki/Linux-Development-Setup

Currently identified (list subject to change, please wait for update)
 - `libtinyxml2-dev`
 - `libasio-dev`

wjwwood, 2017-04-13:
@kenny-y yeah, that's likely related to #332, @nuclearsandwich and @dirk-thomas can confirm whether or not we need that, but it would make sense to me.

Thanks!

nuclearsandwich, 2017-04-13:
@kenny-y the final code block of [Linux Development Setup: How to set up the development environment](https://github.com/ros2/ros2/wiki/Linux-Development-Setup#how-to-setup-the-development-environment) already includes those two packages after the "dependencies for FastRTPS" comment.

We're in the process of confirming that we can drop the Boost packages from the dependency list so you're welcome to try without them. Once they're gone, it should hopefully be a lot clearer with just two FastRTPS-specific dependencies.

kenny-y, 2017-04-13:
Weird, I did followed the command lines and installed all of them, but those 2 were reported missing... then I had to manually install them to make everything work... it could be something wrong in my environment...

I'll close this issue, since the 2 libraries are already listed.

closed 2017-04-13 by kenny-y