ros2/ros2 #338: Unable to build ROS2 on Ubuntu 16.04

davetcoleman, 2017-04-16:
I tried to build from source ROS2 on my 16.04 system that already has ROS1 installed using [these instructions](https://github.com/ros2/ros2/wiki/Linux-Development-Setup). I kept getting errors so I then setup this [Docker file](https://github.com/davetcoleman/ros2_experimental/blob/master/Dockerfile) using the exact ROS2 instructions and I still get the following errors:

```
+++ Building 'connext_cmake_module'
...
CMake Warning at cmake/Modules/FindConnext.cmake:259 (find_package):
  By not providing "Findnddscpp.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "nddscpp", but
  CMake did not find one.

  Could not find a package configuration file provided by "nddscpp" with any
  of the following names:

    nddscppConfig.cmake
    nddscpp-config.cmake

  Add the installation prefix of "nddscpp" to CMAKE_PREFIX_PATH or set
  "nddscpp_DIR" to a directory containing one of the above files.  If
  "nddscpp" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  CMakeLists.txt:9 (find_package)

CMake Warning at cmake/Modules/FindConnext.cmake:260 (find_package):
  By not providing "Findrticonnextmsgcpp.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "rticonnextmsgcpp", but CMake did not find one.

  Could not find a package configuration file provided by "rticonnextmsgcpp"
  with any of the following names:

    rticonnextmsgcppConfig.cmake
    rticonnextmsgcpp-config.cmake

  Add the installation prefix of "rticonnextmsgcpp" to CMAKE_PREFIX_PATH or
  set "rticonnextmsgcpp_DIR" to a directory containing one of the above
  files.  If "rticonnextmsgcpp" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:9 (find_package)
```

```
+++ Building 'opensplice_cmake_module'
Running cmake because arguments have changed.
==> '. /root/ws_ros2/build/opensplice_cmake_module/cmake__build.sh && /usr/bin/cmake /root/ws_ros2/src/ros2/rmw_opensplice/opensplice_cmake_module -DBUILD_TESTING=1 -DAMENT_CMAKE_SYMLINK_INSTALL=1 -DCMAKE_INSTALL_PREFIX=/root/ws_ros2/install' in '/root/ws_ros2/build/opensplice_cmake_module'
-- Found ament_cmake: 0.0.0 (/root/ws_ros2/install/share/ament_cmake/cmake)
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.5.2", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Override CMake install command with custom implementation using symlinks instead of copying resources
CMake Warning at cmake/Modules/FindOpenSplice.cmake:90 (find_package):
  Could not find a package configuration file provided by "opensplice" with
  any of the following names:

    openspliceConfig.cmake
    opensplice-config.cmake

  Add the installation prefix of "opensplice" to CMAKE_PREFIX_PATH or set
  "opensplice_DIR" to a directory containing one of the above files.  If
  "opensplice" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  CMakeLists.txt:8 (find_package)
```

among others

dirk-thomas, 2017-04-16:
If you use a Docker container you likely haven't set a locale. You need to set the locale to any UTF-8 variation (depending on your language): see https://github.com/ros2/ci/blob/87aeb9301903cb63d665c05b35348f3a3cb5ad7f/linux_docker_resources/Dockerfile#L11-L13

dhood, 2017-04-16:
The first two outputs that you listed are just warnings letting you know that optional DDS vendors installations have not been found. These two warnings are safe to ignore if you are not trying to use Connext or Opensplice RMW implementations (respectively). I don't see evidence from your dockerfile that you are interested in using anything other than the default RMW implementation (using Fast-RTPS), so these should be OK to ignore (see [this page](https://github.com/ros2/ros2/wiki/Working-with-multiple-RMW-implementations) for more details).

The third output you listed is indeed an error. I have reproduced it with your dockerfile and believe it to be caused by the default encoding on the dockerfile not being set to utf-8. [These changes](https://github.com/davetcoleman/ros2_experimental/pull/1/files#diff-3254677a7917c6c01f55212f86c57fbfR11) make the dockerfile successfully build.

dhood, 2017-04-18:
@davetcoleman is this right to be closed now?

davetcoleman, 2017-04-18:
The docker builds now with the locale, thanks!

However when building locally (without Docker) on 16.04 I still get:

```
==> '. /home/dave/ros/current/ws_ros2/build/example_interfaces/cmake__build.sh && /usr/bin/cmake /home/dave/ros/current/ws_ros2/src/ros2/examples/example_interfaces -DBUILD_TESTING=1 -DAMENT_CMAKE_SYMLINK_INSTALL=1 -DCMAKE_INSTALL_PREFIX=/home/dave/ros/current/ws_ros2/install' in '/home/dave/ros/current/ws_ros2/build/example_interfaces'
-- Found ament_cmake: 0.0.0 (/home/dave/ros/current/ws_ros2/install/share/ament_cmake/cmake)
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Override CMake install command with custom implementation using symlinks instead of copying resources
CMake Error at CMakeLists.txt:10 (find_package):
  By not providing "Findrosidl_default_generators.cmake" in CMAKE_MODULE_PATH
  this project has asked CMake to find a package configuration file provided
  by "rosidl_default_generators", but CMake did not find one.

  Could not find a package configuration file provided by
  "rosidl_default_generators" with any of the following names:

    rosidl_default_generatorsConfig.cmake
    rosidl_default_generators-config.cmake

  Add the installation prefix of "rosidl_default_generators" to
  CMAKE_PREFIX_PATH or set "rosidl_default_generators_DIR" to a directory
  containing one of the above files.  If "rosidl_default_generators" provides
  a separate development package or SDK, be sure it has been installed.

-- Configuring incomplete, errors occurred!
See also "/home/dave/ros/current/ws_ros2/build/example_interfaces/CMakeFiles/CMakeOutput.log".

<== Command '. /home/dave/ros/current/ws_ros2/build/example_interfaces/cmake__build.sh && /usr/bin/cmake /home/dave/ros/current/ws_ros2/src/ros2/examples/example_interfaces -DBUILD_TESTING=1 -DAMENT_CMAKE_SYMLINK_INSTALL=1 -DCMAKE_INSTALL_PREFIX=/home/dave/ros/current/ws_ros2/install' failed in '/home/dave/ros/current/ws_ros2/build/example_interfaces' with exit code '1'
<== Command '. /home/dave/ros/current/ws_ros2/build/example_interfaces/cmake__build.sh && /usr/bin/cmake /home/dave/ros/current/ws_ros2/src/ros2/examples/example_interfaces -DBUILD_TESTING=1 -DAMENT_CMAKE_SYMLINK_INSTALL=1 -DCMAKE_INSTALL_PREFIX=/home/dave/ros/current/ws_ros2/install' failed in '/home/dave/ros/current/ws_ros2/build/example_interfaces' with exit code '1'
```

Any thoughts?

dhood, 2017-04-19:
Something might have just gotten mixed up if you were starting and stopping the full installation because of the above errors. First thing I would suggest is rebuilding from scratch in a clean directory, if you haven't already (given that it works fine in the container).

mikaelarguedas, 2017-04-26:
@davetcoleman we will close it for now, feel free to comment here if you still have the issue and we'll reopen it

closed 2017-04-26 by mikaelarguedas

fkromer, 2018-02-17:
I have a [similar problem](https://github.com/fkromer/ros2-devenv/issues/1) when I try to build ROS2 in docker with this [Dockerfile](https://github.com/fkromer/ros2-devenv/blob/core/Dockerfile) (kind of copy & paste of the [Linux Development Setup](https://github.com/ros2/ros2/wiki/Linux-Development-Setup) instructions):

```
Step 19/39 : RUN src/ament/ament_tools/scripts/ament.py build --build-tests --symlink-install --isolated src/ros2/ament_cmake_ros/
 ---> Running in 62005d23c04b
# Topological order
 - ament_cmake_ros

Process package 'ament_cmake_ros' with context:
--------------------------------------------------------------------------------
 source_space => /home/ros2/ros2_ws/src/ros2/ament_cmake_ros/ament_cmake_ros
  build_space => /home/ros2/ros2_ws/build_isolated/ament_cmake_ros
install_space => /home/ros2/ros2_ws/install_isolated/ament_cmake_ros
   make_flags => -j4, -l4
  build_tests => True
--------------------------------------------------------------------------------
+++ Building 'ament_cmake_ros'
Running cmake because arguments have changed.
==> '. /home/ros2/ros2_ws/build_isolated/ament_cmake_ros/cmake__build.sh && /usr/bin/cmake /home/ros2/ros2_ws/src/ros2/ament_cmake_ros/ament_cmake_ros -DBUILD_TESTING=1 -DAMENT_CMAKE_SYMLINK_INSTALL=1 -DCMAKE_INSTALL_PREFIX=/home/ros2/ros2_ws/install_isolated/ament_cmake_ros' in '/home/ros2/ros2_ws/build_isolated/ament_cmake_ros'
CMake Error at CMakeLists.txt:5 (find_package):
  By not providing "Findament_cmake.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "ament_cmake", but CMake did not find one.

  Could not find a package configuration file provided by "ament_cmake" with
  any of the following names:

    ament_cmakeConfig.cmake
    ament_cmake-config.cmake

  Add the installation prefix of "ament_cmake" to CMAKE_PREFIX_PATH or set
  "ament_cmake_DIR" to a directory containing one of the above files.  If
  "ament_cmake" provides a separate development package or SDK, be sure it
  has been installed.

-- Configuring incomplete, errors occurred!
See also "/home/ros2/ros2_ws/build_isolated/ament_cmake_ros/CMakeFiles/CMakeOutput.log".

<== Command '. /home/ros2/ros2_ws/build_isolated/ament_cmake_ros/cmake__build.sh && /usr/bin/cmake /home/ros2/ros2_ws/src/ros2/ament_cmake_ros/ament_cmake_ros -DBUILD_TESTING=1 -DAMENT_CMAKE_SYMLINK_INSTALL=1 -DCMAKE_INSTALL_PREFIX=/home/ros2/ros2_ws/install_isolated/ament_cmake_ros' failed in '/home/ros2/ros2_ws/build_isolated/ament_cmake_ros' with exit code '1'
<== Command '. /home/ros2/ros2_ws/build_isolated/ament_cmake_ros/cmake__build.sh && /usr/bin/cmake /home/ros2/ros2_ws/src/ros2/ament_cmake_ros/ament_cmake_ros -DBUILD_TESTING=1 -DAMENT_CMAKE_SYMLINK_INSTALL=1 -DCMAKE_INSTALL_PREFIX=/home/ros2/ros2_ws/install_isolated/ament_cmake_ros' failed in '/home/ros2/ros2_ws/build_isolated/ament_cmake_ros' with exit code '1'
The command '/bin/sh -c src/ament/ament_tools/scripts/ament.py build --build-tests --symlink-install --isolated src/ros2/ament_cmake_ros/' returned a non-zero code: 1
```

mikaelarguedas, 2018-02-17:
@fkromer This build command doesnt match the instructions, you are trying to build only the `ament_cmake_ros` package (as you path the path to only that package to ament build).
The issue you are facing is that you didnt build (or source the resulting workspace) of `ament_cmake_ros` dependencies so CMake cannot find them.

The recommended way to build only one package of the workspace (assuming all its dependencies have been built) is to use the `--only-package` flag rather than passing a custom workspace path: e.g.
`RUN src/ament/ament_tools/scripts/ament.py build --build-tests --symlink-install --isolated 
 --only-package ament_cmake_ros`
If you want to build all it's dependencies as well you can build all packages "up to" your package by using the `--end-with` flag:
`RUN src/ament/ament_tools/scripts/ament.py build --build-tests --symlink-install --isolated 
 --end-with ament_cmake_ros`

Relevant Docker resources:
-----------------------------------
If you want to build ROS 2 from source in a container, you may find this useful: https://github.com/osrf/docker_images/blob/master/ros2/source/source/Dockerfile (if you want to build master instead of the ardent release you can replace `release-latest` with `master` here https://github.com/osrf/docker_images/blob/cf42123ace19919ff2ea66a712884a32abbadf31/ros2/source/source/Dockerfile#L82)

fkromer, 2018-02-17:
> This build command doesnt match the instructions, you are trying to build only the ament_cmake_ros package (as you path the path to only that package to ament build).
The issue you are facing is that you didnt build (or source the resulting workspace) of ament_cmake_ros dependencies so CMake cannot find them.

@mikaelarguedas Thanks a lot for the explanation.

> Relevant Docker resources:
>
> If you want to build ROS 2 from source in a container, you may find this useful: https://github.com/osrf/docker_images/blob/master/ros2/source/source/Dockerfile (if you [...]

That's exactly what I was loooking for.

TheLaas, 2019-05-08:
Sorry to open this thread again, but I have a similar problem and I can't get the suggested solutions to work. Building from within an interactive docker bash with `colcon build --base-paths DummyPublisher/` works fine but trying to do the same in the dockerfile generates an error. I'm new to both Ros and docker so I might have missed something trivial. This is my dockerfile:
```
FROM osrf/ros2:nightly

RUN apt-get update && apt-get install vim -y

RUN apt-get install -q -y \
    libasio-dev \
    libtinyxml2-dev \
    wget \
    && rm -rf /var/lib/apt/lists/*

COPY . /usr/src/project
WORKDIR /usr/src/project
RUN mkdir src
RUN wget https://raw.githubusercontent.com/ros2/ros2/crystal/ros2.repos \
    && vcs import src < ros2.repos

RUN apt-get update && rosdep install -y \
    --from-paths src \
    --ignore-src \
    --skip-keys "console_bridge fastcdr fastrtps libopensplice67 libopensplice69 rti-connext-dds-5.3.1 urdfdom_headers" \
    && rm -rf /var/lib/apt/lists/*

RUN colcon build --base-paths DummyPublisher/

LABEL Name=dnse-fsn Version=0.0.1
```
The CMakeLists.txt:
```
cmake_minimum_required(VERSION 3.5)
project(dummy_publisher)

if(NOT CMAKE_CXX_STANDARD)
  set(CMAKE_CXX_STANDARD 14)
endif()

if(CMAKE_COMPILER_IS_GNUCXX OR CMAKE_CXX_COMPILER_ID MATCHES "Clang")
  add_compile_options(-Wall -Wextra -Wpedantic)
endif()

find_package(ament_cmake REQUIRED)
find_package(rclcpp REQUIRED)
find_package(std_msgs REQUIRED)

add_executable(dummy_publisher DummyPublisher.cpp)
ament_target_dependencies(dummy_publisher rclcpp std_msgs)

install(TARGETS
  dummy_publisher
  DESTINATION lib/${PROJECT_NAME}
)

ament_package()
```
and the error I get when I try to run this:
```
--- stderr: dummy_publisher
CMake Error at CMakeLists.txt:12 (find_package):
  By not providing "Findament_cmake.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "ament_cmake", but CMake did not find one.

  Could not find a package configuration file provided by "ament_cmake" with
  any of the following names:

    ament_cmakeConfig.cmake
    ament_cmake-config.cmake

  Add the installation prefix of "ament_cmake" to CMAKE_PREFIX_PATH or set
  "ament_cmake_DIR" to a directory containing one of the above files.  If
  "ament_cmake" provides a separate development package or SDK, be sure it
  has been installed.

---
Failed   <<< dummy_publisher    [ Exited with code 1 ]
```

dirk-thomas, 2019-05-08:
@TheLaas your package `dummy_publisher` tries to use `ament_cmake` but likely doesn't declare that dependency in the `package.xml` file which you haven't shared.

mikaelarguedas, 2019-05-08:
@TheLaas ROS provides setup files to configure your environment to be able to find ROS packages. You need to source the setup file before building, you can see https://github.com/mikaelarguedas/ros2ci/blob/baa9ca1bf488d8d2425301979ec38d3f25947c0d/Dockerfile#L43 for an example

If you enter the container and then run your `colcon build` command it will work as the entry point of the container is sourcing the setup file for you.

Please consider asking your questions on https://answers.ros.org as it will have more visibility than commenting on a closed issue. There may also be questions already answered by the community that can help you: https://answers.ros.org/question/319610/create-dockerfile-for-ros2-package-ament_cmake-error/