ros2/ros2 #339: DDS Security support

mikaelarguedas, 2017-04-18:
meta ticket to aggregate PRs

mikaelarguedas, 2017-05-27:
first security tests on all platforms :crossed_fingers: (code will still need to be cleaned up a bit before review)
* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

mikaelarguedas, 2017-05-30:
Placing this in review, I'll comment on individual PRs to point to where I'm looking for advice. Most of it should be pretty digestible, only the system_tests one will be a bit painful to review

dirk-thomas, 2017-06-01:
Overall the PRs look good. I made a few small comments.

One thing I am concerned about is how deeply the security feature is embedded into the API. Imo it would be preferred if the codebase could be build without using anything security related. Yes, it can be "not used" and an rmw provider can simply return errors but I was more thinking about not being part of the API when not being used. In the DDS spec that is basically achieved by making the security specific stuff part of the the dynamic QoS flags. (Anyway this is not an argument to delay merging this as is. It is more trying to formulate the vision I think this should be going to in the future to avoid this tight coupling.)

mikaelarguedas, 2017-06-06:
> Yes, it can be "not used" and an rmw provider can simply return errors but I was more thinking about not being part of the API when not being used.

That is a good point. I think we can revisit this once we want to provide an extensible / pluggable option system to node participant / creation. I will not target this in this set of PRs but that's definitely something that we should keep in mind in the future.

mikaelarguedas, 2017-06-15:
another round of CI now that every branch has been approved and rebased:
* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

mikaelarguedas, 2017-06-15:
:green_apple: :green_book: :green_heart: :green_heart: 
Merging

wjwwood, 2017-06-15:
@mikaelarguedas, so sorry to say some new nightlies are now unstable and it looks related to the security changes 😧:

http://ci.ros2.org/job/nightly_win_rel/463/testReport/

mikaelarguedas, 2017-06-15:
:confounded: Forgot to deploy the changes to CI so the job were not reconfigured to pass -DSECURITY=ON. I'll add an option so that the security tests are ran only if this option is passed. Thanks for letting me know

mikaelarguedas, 2017-06-15:
Deployed. I noticed that the packaging jobs don't use a the same default arguments so they won't use the -DSECURITY options just yet. I'll open a follow up PR

dirk-thomas, 2017-06-15:
The current cmake flags don't work when passing some flags behind them: http://ci.ros2.org/job/ci_osx/2204/console

mikaelarguedas, 2017-06-15:
Reported on the appropriate issue [here](https://github.com/ros2/ci/issues/51#issuecomment-308776348)

mikaelarguedas, 2017-06-17:
Evey PR landed :tada: . Closing this

closed 2017-06-17 by mikaelarguedas