ros2/ros2 #342: Cartographer support

clalancette, 2017-04-19:
This is a meta-ticket for Cartographer support in ROS2.

mikaelarguedas, 2017-06-14:
@clalancette Can this ticket be closed ?

mikaelarguedas, 2017-06-14:
Yeah I don't think we should keep a ticket open for every package we want to improve. Specific issues about what to improve is more relevant IMO. (it's also to cleanup the waffleboard because my screen is getting too small for this ticket)

mikaelarguedas, 2017-06-14:
> (like ros2/teleop_twist_joy#3, etc)

This is not related to cartographer and will be useful for any demo using teleop of any robot

clalancette, 2017-06-14:
Heh, OK.  Closing it out, then.

closed 2017-06-14 by clalancette

YuehChuan, 2017-06-22:
HI, wonder is there some details  about how to  make ROS2  on Cartographer? TQ!

clalancette, 2017-06-22:
@YuehChuan We're still in the process of writing the documentation for it.  Once that is done, it should be available in the https://github.com/ros2/turtlebot2_demo repository (or, at least, linked to from there).

mikaelarguedas, 2017-06-22:
Hi @YuehChuan, we will give instructions of how to build and run the cartographer demo on [this repository](https://github.com/ros2/turtlebot2_demo). The instructions will be available in the next week or 2

YuehChuan, 2017-08-12:
@clalancette , @mikaelarguedas ,Thanks for the info! I just notice the reply now. Now I'm compiling ros2 beta2 on raspberry pi3.