ros2/ros2 #344: Sometimes messages are lost

mc-android-developer, 2017-04-21:
I have two different computers in same network both running Ubuntu. Each runs ROS2 node.
One node is subscribed for the topic "testtopic" another is publishing to it.
Publisher code is:
<pre>
    auto node = std::make_shared&lt;Node&gt;("testnode");
    rclcpp::executors::SingleThreadedExecutor controllerExecutor;
    controllerExecutor.add_node(node);

    auto spinLambda= [&controllerExecutor]() {
        controllerExecutor.spin();
    };
    std::thread nodeExecutionThread(spinLambda);
    // sleep_for(1s);

    static auto pub = node-&gt;create_publisher&lt;std_msgs::msg::String&gt;("testtopic", rmw_qos_profile_default);
    auto msg = std::make_shared&lt;std_msgs::msg::String&gt;();
    msg-&gt;data = "testmsg";
    pub-&gt;publish(msg);
    nodeExecutionThread.join();
</pre>

Problem: messages are sometimes lost.

I was able to make it work all the time by adding sleep_for(1s) in the 9th line.
When sleep_for(1s) is commented out, messages are lost. Especially if I run the code multiple times. I also checked network traffic using Wireshark and I discovered that RTPS frames are not actually sent.
If I uncomment sleep_for(1s), all messages are sent normally and I can see matching RTPS frames in Wireshark.

mc-android-developer, 2017-04-22:
Maybe it has something to do with multithreading. Do spinning and publishing have to be done in the same thread?

wjwwood, 2017-04-22:
> Maybe it has something to do with multithreading. Do spinning and publishing have to be done in the same thread?

No, spinning and publishing do not need to happen in the same thread. In fact, at the moment at least, you don't need to spin to publish. You only have to spin for timers, subscriptions, and services (both client and server).

Sorry I haven't had time to read up on your original post, I'll get to it asap.

wjwwood, 2017-04-28:
Sorry, I have a few more questions for you:

- Which version of ROS 2 are you using (from source or a binary from something like beta1)?
- Is the subscriber running all of the time or are you starting and stopping it too?

wjwwood, 2017-04-28:
I can reproduce what you're talking about. Even if I just modify our `talker` to publish once, I often do not see it make it to the `listener`.

This could be a bug in the middleware (I'm assuming Fast-RTPS in your case, that's what I used and I haven't tried Connext yet). I think we should continue to follow up on and see if we can confirm why it is happening.

On the other hand, I believe it could also just come down to the way discovery works in DDS. Because the discovery happen asynchronously, you could be getting to your publish call before the publisher and subscriber find one another. That would support why adding the sleep where you did might help. I am a little surprised that you don't need to put the sleep between the `create_publisher` and `publish` calls to make it work better. This implies it is actually the "participant" discovery that is causing the race, and not necessarily the "end point" discovery.

Usually, even in ROS 1, if you have a program that publishes once and never again you latch it. I tried the equivalent in my testing and it actually didn't seem to help. Which makes me think that both the idea of a discovery race issue and/or a bug in the middleware are more likely.

mc-android-developer, 2017-04-28:
Thank you for response.

We compiled ROS2 from sources on both machines as described here
https://github.com/ros2/ros2/wiki/Linux-Development-Setup

Version from ros2.repos file is 
version: 148df7e841b91da488ec3bb4abd295a2bccdb728

When I was experimenting I ran subscriber once and it was running all the time without restart.
In fact when I realized the problem, I started using regular 'listener' demo app (from src/ros2/demos/demo_nodes_cpp/src/topics/) as a subscriber with the topic changed to 'testtopic', but it didn't help.

wjwwood, 2017-06-06:
I don't have any more info for this issue, but I thought I'd wrap it up for now with a few things that someone could look into if they have the time:

- See if this issue exists with Connext
- Watch what is happening in Wireshark to see if discovery packets are being resent or acknowledged
- Try to turn on a deeper level of debug output for FastRTPS
- Try to reproduce with pre FastRTPS code (to exclude our ROS 2 code as the issue)

realdream, 2018-06-07:
I got similar issue on multi machine through LAN or WIFI

### Required Info for Both machine:
Operating System:
Ubuntu 16.04
Installation type:
From source
Version or commit hash:
Both Current master branch and ardent release
DDS implementation:
Fast-RTPS
Client library (if applicable):
both rclcpp rclpy

### Steps to reproduce issue
on first machine:
```
ros2 run demo_nodes_cpp listener
```
on second machine:
```
1) ros2 run demo_nodes_cpp listener
2) then use Ctrl-C to interrupt 
3) repeat step 1 2 for multiple times 
```
you will notice that the listener will not get msg from time to time

realdream, 2018-06-07:
# more Information to add:
1 When I turn on the debug info for Faster-RTSP , I'm not able to reproduce the issue.
2 When delay is added between 'create_publisher' and 'publish ' , I'm not able to reproduce the issue.

dirk-thomas, 2018-07-05:
Can you please test this again with the recent ROS 2 Bouncy release and comment with the results. Thanks.

mcsheehan, 2018-11-22:
@dirk-thomas seeing as the unreliable messaging was one of the main reasons for creating ros2 in the first place I think you should have automated integration tests left right and center on your publisher and subscribers. Otherwise you are offloading this responsibility to every other node on your event bus, and have no guarantees that your event bus is even working properly with each code change.

mc-android-developer, 2018-11-27:
@dirk-thomas Sorry for delay
I've done a number of tests today. I pulled release-latest and built it from sources as explained here https://index.ros.org/doc/ros2/Linux-Development-Setup/. After that I used listener/talker demo apps. Each test was run 5-7 times.
1. Run listener/talker on the same machine. Everything works. No lost messages.
2. Run listener/talker on different machines in the same network. Everything works. No lost messages.

After that I changed talker to reduce delay between messages. I put 10ms instead of 1s for wall timer here src/ros2/demos/demo_nodes_cpp/src/topics/talker.cpp:Ln68

3. Run listener/talker on the same machine. Everything works. No lost messages.
4. Run listener/talker on different machines in the same network. Usually from 3 to 15 first messages are stably lost. When listener start receiving after first loss, no more lost messages. Problem only happens with a few dozens of first messages. I also tried to use default QOS in talker. Result is the same.

After that I reduced delay even more. I put 1ms instead of 10ms.

5. Run listener/talker on the same machine. Everything works. No lost messages.
6. Run listener/talker on different machines in the same network. Usually from 50 to 150 first messages are stably lost.

All in all, it fells way better now. Are cases #4 and #76 an issue or it’s expected for high frequency?

dirk-thomas, 2019-01-22:
@mc-android-developer I would think this is expected behavior. The talker starts publishing messages right away and doesn't wait for any subscriber to be there. It will take some non-zero time for the subscriber to be discovered and only after that messages are being received by the subscriber. So increasing the frequency of published messages will drop more messages before that is the case.

If you don't want that kind of behavior you:

* either have to change the QoS settings to store all published messages and deliver them to laterer joining subscribers (which can require a lot of memory to store and a lot of bandwidth to send all cached message to new subscribers)
* or you can wait in your publisher until a certain number of expected subscribers has been matched (though that undermines the idea of anonymous publish/subscribe a little).

dirk-thomas, 2019-01-22:
I will go ahead and close this ticket for now since nothing in this thread seems to indicate a bug in ROS 2. Please feel free to comment with more information and the ticket can be reopened if necessary.

closed 2019-01-22 by dirk-thomas

reza-ebrahimi, 2021-02-15:
I'm confirming this message lost problem, Even in the case of spinning `forever`, sometimes messages are lost.

fujitatomoya, 2021-02-15:
@reza-ebrahimi it would be probably nice to open new issue with description about reproducible procedure. i tried with https://github.com/ros2/ros2/issues/344#issuecomment-395332842, but i cannot reproduce the issue.

rfpg1, 2024-04-13:
Is there any fix to this? I still can reproduce this in 2024

fujitatomoya, 2024-04-13:
@rfpg1 i would suggest that probably you can create another issue with description how to reproduce the issue and distro versions. this issue has been closed more than 5 years.