ros2/ros2 #350: ROS Time and use_sim_time

wjwwood, 2017-05-09:
This issue will aggregate the work on a ROS Time object in the client libraries (C++ and Python) as well as implementing the simulated time similar to (if not exactly like) `roscpp::Time` + `use_sim_time` from ROS 1.

----

There is already a design doc about ROS Time here: http://design.ros2.org/articles/clock_and_time.html

This document describes use cases for having a time abstraction, rather than using system or language time interfaces directly. It also describes how we will implement something very similar to how `/clock` works in ROS 1 (at least for now) and that the "ROS Time" object will need to be created with, or gotten from, a Node object.

It also states that we will control when simulated time is turned on the same as with ROS 1, via a parameter called something like `use_sime_time`. Except in ROS 2, that parameter will be on each Node. It is an open question on how to make it easy to change that value globally, perhaps by also having nodes look at a "global" version of `use_sim_time` on a "global parameter server" node if present.

It was also decided, for now, that if a "ROS Time" object is asked the time before the node has created the subscription to `/clock` or before any data is received from `/clock` then it will return `0` (or the equivalent time point) until data is received. This is consistent with how ROS 1 does it, but I would be happy to entertain other possibilities.

----

TODO: Sketch of the API to placed here when we agree on it.

----

Things to do:

- remove `rcl_time_source_init()`
  - rationale: since the "ros time" source needs to be initialized with a rcl_node_t as input, this method cannot serve all of the various time sources
  - i.e. the construction of a system time source is different from a ros time source and so they cannot share a common constructor
- Finish the `rclcpp::Time` (or whatever it will be called) object
- Make `rclcpp::Time::now()` no longer be static
- TODO more things here I'm sure...

wjwwood, 2017-05-09:
I created this issue based on discussion here:

sadfasdf

@REPLACEME fyi

----

When we decide on an API sketch I'll put it in the original post.

For the sketch of the API, I started with this for C++ (incomplete):

```c++
class rclcpp::BaseTime
{
public:
  virtual uint64_t nanoseconds();
  virtual void from_nanoseconds(uint64_t);
  virtual double seconds();
  virtual void from_seconds(double);
private:
  rcl_time_point_t rcl_time_point_;
};

class rclcpp::SystemTime : public rclcpp::BaseTime
{
public:
  static rclcpp::SystemTime now();
  template <typename DurationT = std::chrono::system_clock::duration>
  virtual std::chrono::time_point<std::chrono::system_clock, DurationT> as_std_time_point();
};

class rclcpp::SteadyTime : public rclcpp::BaseTime
{
public:
  template <typename DurationT = std::chrono::steady_clock::duration>
  virtual std::chrono::time_point<std::chrono::steady_clock, DurationT> as_std_time_point();
};

class rclcpp::Time : public rclcpp::BaseTime  // "ROS Time"
{
public:
  Time() = delete;
  Time(rclcpp::Node::SharedPtr node);

  template <typename DurationT = std::chrono::system_clock::duration>
  virtual std::chrono::time_point<std::chrono::system_clock, DurationT> as_std_time_point();
};

rclcpp::Time rclcpp::Node::get_time_object();  // rclcpp::Time factory method, but not static
```

But I began to think that we don't really need to make our own System and Steady time objects in C++ since the signatures looked so much like the "clocks" in `std::chrono`.

So to everyone out there:

- Should we have our own C++ objects for System time and Steady time?

For me, to answer that I'd have to see more mock API usage to see if it is too cumbersome to use `std::chrono` directly when not using our "ROS Time" object. I'll try to sketch up that API too and see what that looks like. Anyone else that has ideas for this interface are welcome too.

wjwwood, 2017-05-09:
/cc @BrannonKing

BrannonKing, 2017-05-09:
I don't care for idea of a "global parameter server". I think we should leave that behind in ROS1; run-time-changable settings should be very rare and people can write their own messages for those extreme cases. Everyone uses scripts to start their nodes. It's simple to pass command line parameters to nodes. It's also simple to use environment variables. I think we should support those two parameter methods (both used at startup) rather than any kind of run-time parameters for enabling "ROS Time".

BrannonKing, 2017-05-09:
Your `BaseTime` interface leaves out a conversion to `builtin-interfaces::Time`. You want a static method for that conversion? An implicit cast? Where should it live?

wjwwood, 2017-05-13:
> I don't care for idea of a "global parameter server". I think we should leave that behind in ROS1; run-time-changable settings should be very rare and people can write their own messages for those extreme cases.

I think you're mixing up two things. We've made all of our parameters node specific now, so there's no concept of a "global parameter" any longer. The "global parameter server" I mentioned would just be a node with a well known name that would accept and store any parameter given to it.

Separately there is the issue of changing parameters at runtime vs during runtime. During runtime is facilitated with something called `dynamic_reconfigure` in ROS 1, but we're building that feature into the normal parameters of ROS 2. You can turn off this feature (so it is optional), but I think it's an extremely useful pattern, especially for things like device drivers or algorithms during R&D.

> Everyone uses scripts to start their nodes. It's simple to pass command line parameters to nodes.

Just but what ever system we provide for launching nodes will need to make this easier, because it is very cumbersome to pass an extra parameter to every node, rather than set a single parameter with one line in a launch file like ROS 1. It could be that setting this parameter on every node that is launched is the right way to do it, but we just have to make sure it's easy to do.

> It's also simple to use environment variables.

The problem with environment variables is that they are not as introspectable as ROS parameters. If I'm not a workstation and I want to know whether or not a node on a robot computer is using sim time, then an environment variable is a difficult thing to check for, especially while it is running.

An environment variable might be fine for seeding the value in the parameter, but I wouldn't make it the only way to control it.

> I think we should support those two parameter methods (both used at startup) rather than any kind of run-time parameters for enabling "ROS Time".

I think using a ROS parameter + an environment variable is good enough. But if in the future we setup a general purpose global parameter server node (which would always be optional) it can be one of the options for seeding the value. You can imagine a line of succession like this:

- Start a ROS Parameter called `use_sim_time` which is `= false`
- If the global parameter server is available and `use_sim_time` is set, store that value
- If the environment variable is set, store that value
- If the `use_sim_time` was passed as a command line/startup argument to the node, store that value
- Start the ROS Time subsystem with whatever ended up in `use_sim_time`

This isn't necessary, but I'm trying to say they're not mutually exclusive. The important thing is the result ends up in a ROS parameter so it can be introspected as well as recorded by rosbag.

Whether or not we'll need to support changing of the parameter _during_ runtime is a different question. Off-hand I would think it shouldn't be supported, but I might be wrong. We can start with that for now and try to come up with a different policy if a use case comes up.

> We should probably define some naming convention for ROS2 environment variables and command line parameters (used by the core). This is to avoid conflicts with other software packages or user-defined command line parameters.

This is will probably be covered in a new design document at some point, but they must also coexist with the remapping command line arguments, which has been considered in the design of the remapping syntax:

http://design.ros2.org/articles/static_remapping.html

It's also worth remembering that not all nodes will be run as a single process which can take command line arguments. In fact our hope is that nodes will not have their own main (and so their own command line arguments) in most cases. So that they can be reused and combined with other nodes into single processes through composition.

> Your `BaseTime` interface leaves out a conversion to `builtin_interfaces::msg::Time`. You want a static method for that conversion? An implicit cast? Where should it live?

Yeah, I stopped short of that when I couldn't decide on whether or not we needed a `BaseTime` class or if we should just make a ROS Time class which mimics the `std::chrono` clocks.

It might be that we can just setup a conversion constructor or overload the `=` operator in a free function or something like that (I haven't thought through that completely), but at the very least we can have a non-static `to_time_msg` function as a member of the ROS Time class, even if that's not the most convenient API.

tfoote, 2018-04-26:
@wjwwood Can you reconcile this with the state in #178?

wjwwood, 2018-10-10:
I think I'll just close this, allowing it to be superseded by #178, but it would have been nice to build off the existing work...

closed 2018-10-10 by wjwwood

JWCS, 2023-01-10:
I'm struggling to find this, but is there a way to globally declare use_sim_time, or direct nodes to listen to /clock?
The best I can find is hacky solutions involving dozens of lines of arg/param passing, which is very fragile.

fujitatomoya, 2023-01-13:
I think it can be only controlled by Node Parameter `use_sim_time` currently.

what you are looking for is like global switch that all nodes use simulation time, right? e.g) environmental variable to set, so that all nodes instantiated by that shell process can start the nodes use simulation time?