ros2/ros2 #353: include debug libraries for OpenCV

dirk-thomas, 2017-05-15:
In the archive https://github.com/ros2/ros2/releases/download/release-alpha1/opencv-2.4.11-win-vs2015-x64.zip

Once the debug libraries are being used the workaround from ros2/vision_opencv#6 can be reverted.

mikaelarguedas, 2017-06-24:
This has the potential of fixing the `cam2image` and `showimage` test faillures on windows debug (they consistently fail on all the machines)

mikaelarguedas, 2017-07-25:
I can confirm that both the `image_geometry` and the `image_tools` tests pass once openCV debug libraries installed:
* Windows [image]

Though not sure what the best way to release them alongside the optimized libraries is yet

dirk-thomas, 2017-07-25:
The debug libraries should be shipped along side the release libraries with a debug suffix. The CMake module should then export both of them using the proper keywords (`optimized` / `debug`). For FastRTPS we are doing something similar: https://github.com/ros2/rmw_fastrtps/blob/c0f148ee03156860492b220c66358b8af519b095/fastrtps_cmake_module/cmake/Modules/FindFastRTPS.cmake#L45-L92

mikaelarguedas, 2017-07-25:
Yes I agree with that :+1: thanks for the pointer, I just haven't looked into it in enough details to decide if it's simpler to modify their generated INSTALL target VS solution or if we should make our own CMake logic to override their generated CMake files.

mikaelarguedas, 2017-07-26:
Ok actually You can build their "INSTALLATION" target in both Release and Debug mode. the artifact is [here](https://github.com/ros2/ros2/releases/download/release-beta2/opencv-2.4.13.2-vc14.VS2015.zip). It has been deployed on icecube. I just launched
* Windows [image] to test the new artifact. 
If it passes I'll propagate it to all windows machines for the nightlies. And update the installation instructions accordingly if all is green tomorrow

mikaelarguedas, 2017-07-26:
All windows nodes now have the new opencv and the installation instructions have been updated accordingly on the wiki. Closing this

closed 2017-07-26 by mikaelarguedas