ros2/ros2 #354: fix flaky lifecycle tests

Karsten1987, 2017-05-23:
top level issue tackling the failing flaky tests on linux.

Connected PRs:
 * rcl: ros2/rcl#139
 * rclcpp: ros2/rclcpp#327

CI jobs 
`ament_test_args:  --only rcl_lifecycle rclcpp_lifecycle --retest-until-fail 20 --ctest-args -LE linter -`:
[image]
[image]
[image]

mikaelarguedas, 2017-05-24:
Looks like one of the tests started failing along the way: http://ci.ros2.org/job/ci_windows/2747/#showFailuresLink

Karsten1987, 2017-05-24:
this should be fixed by now.

Karsten1987, 2017-05-26:
I am going to merge in order to proceed with https://github.com/ros2/rclcpp/issues/329

closed 2017-05-26 by Karsten1987