ros2/ros2 #355: call C:\dev\ros2\local_setup.bat

Yingzhong262526, 2017-06-07:
Hi!
   everyone,when I install ros2 in windows10 from the pre-built binary !run a powershell/cmd  as administrator,I call C:\dev\ros2\local_setup.bat ,there exists some errors as flowing:
call : 无法将“call”项识别为 cmdlet、函数、脚本文件或可运行程序的名称。请检查名称的拼写，如果包括路径，请确保路径正确
，然后再试一次。
所在位置 行:1 字符: 1
+ call C:\dev\ros2\local_setup.bat
+ ~~~~
    + CategoryInfo          : ObjectNotFound: (call:String) [], CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException
why?expect yours reply!

wjwwood, 2017-06-07:
Try using cmd.exe rather than powershell.

Yingzhong262526, 2017-06-07:
if I use cmd.exe ,the error is : the system can't find the specified path

wjwwood, 2017-06-07:
> if I use cmd.exe ,the error is : the system can't find the specified path

That's not enough information to help you, unfortunately. Please post more steps that you took and more of the error output.

Yingzhong262526, 2017-06-07:
I flow this install instructions which install ros beta-1 from pre-built binary forms in the win 10 system,the link is https://github.com/ros2/ros2/wiki/Windows-Install-Binary .Step by step , after I  download the ros 2 and unpack it in the C:\dev\ros2 folder, then I try some exmaples whether the ros 2 works on my win10 systems. When I run the command: call C:\dev\ros2\local_setup.bat (the path is correct),if my shell is cmd.exe,the error is:
the system can't find the specified path!

If my shell is powershell,the error is :
call : 无法将“call”项识别为 cmdlet、函数、脚本文件或可运行程序的名称。请检查名称的拼写，如果包括路径，请确保路径正确
，然后再试一次。
所在位置 行:1 字符: 1
+ call C:\dev\ros2\local_setup.bat
+ ~~~~
    + CategoryInfo          : ObjectNotFound: (call:String) [], CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException

I am a beginner of the ros2.0 ,Sincerely hope you can help me ! Thanks a lot first!

wjwwood, 2017-06-07:
It's possible the Python version (and therefore path) changed since the beta 1 release. If you look in the `local_setup.bat` there is a hard coded path to a Python interpreter. For the beta 1 binaries it is expecting to find python at `C:\ProgramData\chocolatey\lib\python3\tools\python.exe`. You can see where your python interpreter is by doing `where python`. And if your path is different then you can override the hard coded path with the `AMENT_PYTHON_EXECUTABLE` environment variable.

You can try taking the output of `where python` and setting the env variable to that like this: `set "AMENT_PYTHON_EXECUTABLE=C:\path\to\python"`. And then try calling the bat file again.

wjwwood, 2017-06-07:
You can also try setting `AMENT_TRACE_SETUP_FILES` to `1` and the setup file will give you more information about what it is doing.

Yingzhong262526, 2017-06-07:
Thank you very much ,now It works!

dirk-thomas, 2017-06-07:
Closing in favor of ros2/build_cop#23.

closed 2017-06-07 by dirk-thomas