ros2/ros2 #356: Ament Tutorial: Windows Fails Symlink

mrutyu1987, 2017-06-07:
Trying tutorial to build Ament on Windows. Getting following Symlink error: Any ideas how resolve this?

```
PostBuildEvent:
  setlocal
  "C:\Program Files\CMake\bin\cmake.exe" -DBUILD_TYPE=Release -P cmake_install.cmake
  if %errorlevel% neq 0 goto :cmEnd
  :cmEnd
  endlocal & call :cmErrorLevel %errorlevel% & goto :cmDone
  :cmErrorLevel
  exit /b %1
  :cmDone
  if %errorlevel% neq 0 goto :VCEnd
  :VCEnd
  -- Install configuration: "Release"
  -- Execute custom install script
  -- Symlinking: D:/ROS/ros2_ws/install/share/ament_index/resource_index/package_run_dependencies/ament_cmake_core
  failed to create symbolic link 'D:/ROS/ros2_ws/install/share/ament_index/resource_index/package_run_dependencies/ament_cmake_core': No error
  CMake Error at ament_cmake_symlink_install/ament_cmake_symlink_install.cmake:293 (message):
    Could not create symlink
    'D:/ROS/ros2_ws/install/share/ament_index/resource_index/package_run_dependencies/ament_cmake_core'
    pointing to
    'D:/ROS/ros2_ws/build/ament_cmake_core/ament_cmake_index/share/ament_index/resource_index/package_run_dependencies/ament_cmake_core'
  Call Stack (most recent call first):
    ament_cmake_symlink_install/ament_cmake_symlink_install.cmake:149 (_ament_cmake_symlink_install_create_symlink)
    ament_cmake_symlink_install/ament_cmake_symlink_install.cmake:305 (ament_cmake_symlink_install_files)
    cmake_install.cmake:31 (include)

```

mikaelarguedas, 2017-06-07:
Thanks for reporting it. IIRC Windows requires administrator rights to symlink files. Regular users can only symlink folders. We should highlight in the ament tutorial this limitation of symlinks on Windows.

mrutyu1987, 2017-06-07:
@mikaelarguedas yes I am building it with Admin rights. Also tried without the --symlink-install option but still same error.

mikaelarguedas, 2017-06-07:
> yes I am building it with Admin rights

Oh interesting I actually haven't tried it with admin priviledges, I will try to do the same and report here

> Also tried without the --symlink-install option but still same error.

did you delete your build and install folder before rerunning your build?

mikaelarguedas, 2017-06-07:
Follow-up discussion on the following discourse thread: https://discourse.ros.org/t/ament-tutorial-build-fails-with-symlink/1899/3

tfoote, 2017-06-07:
@mrutyu1987 Please only start threads in one place. It's poor etiquette to start multiple duplicate threads as you can get multiple people duplicating efforts to help you.  http://wiki.ros.org/Support#Etiquette

mrutyu1987, 2017-06-07:
@tfoote Sorry my mistake. I thought both are different

@mikaelarguedas after deleting the directories the build continued...but now stopped with error:
```
   
  FinalizeBuildStatus:
         Deleting file "test_memory_tools.dir\Release\test_mem.A29DCFE4.tlog\unsuccessfulbuild".
         Touching "test_memory_tools.dir\Release\test_mem.A29DCFE4.tlog\test_memory_tools.lastbuildstate".
    34>Done Building Project "D:\ROS\ros2_ws\build\rcutils\test\memory_tools\test_memory_tools.vcxproj" (default targets).
    20>Done Building Project "D:\ROS\ros2_ws\build\rcutils\test\memory_tools\test_memory_tools.vcxproj.metaproj" (default targets).
     2>Done Building Project "D:\ROS\ros2_ws\build\rcutils\ALL_BUILD.vcxproj.metaproj" (default targets) -- FAILED.
     1>Done Building Project "D:\ROS\ros2_ws\build\rcutils\rcutils.sln" (default targets) -- FAILED.

Build FAILED.

       "D:\ROS\ros2_ws\build\rcutils\rcutils.sln" (default target) (1) ->
       "D:\ROS\ros2_ws\build\rcutils\rcutils.vcxproj.metaproj" (default target) (7) ->
       "D:\ROS\ros2_ws\build\rcutils\rcutils.vcxproj" (default target) (29) ->
       (CustomBuild target) ->
         C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(171,5): error MSB6006: "cmd.exe" exited with code
       1. [D:\ROS\ros2_ws\build\rcutils\rcutils.vcxproj]
```
```

mikaelarguedas, 2017-06-07:
Ok so this is an issue not related to symlinks. Looks like Empy is not found. Did you install all the dependencies listed [here](https://github.com/ros2/ros2/wiki/Windows-Development-Setup#installing-a-few-dependencies)? (more specifically this line ```> pip install EmPy```)

closed 2017-08-08 by mrutyu1987