ros2/ros2 #357: dummy robot demo

Yingzhong262526, 2017-06-08:
Hi:
   Everyone!I am using ros2 on the ubuntu 16.04 to run the demo of "Use the robot state publisher to publish joint states and TF" from the link https://github.com/ros2/ros2/wiki/Tutorials. When you click it link,then it jumps to the link https://github.com/ros2/ros2/wiki/dummy-robot-demo,When I open Terminal 3 and run "python3 ~/ros2_ws/install/share/dummy_robot_bringup/launch/dummy_robot_bringup.py" 
the error occurs:
python3: can't open file '/home/zhong/ros2_ws/install/share/dummy_robot_bringup/launch/dummy_robot_bringup.py': [Errno 2] No such file or directory

I check that path，it reallly can't find the package of dummy_robot_bringup in that path !How can I obtain the package? or some probelms in the installing ros2 from source ?

Thanks a lot first!

mikaelarguedas, 2017-06-08:
Hi @Yingzhong262526,

Thanks for reporting this issue. If I understand correctly you installed ROS2 from source following [these instructions](https://github.com/ros2/ros2/wiki/Linux-Development-Setup) is that correct?

The dummy_robot_demo has been added after the beta1 release. To run it I would suggest you to use the latest version of the code. To do so you can setup a new workspace by following [these instructions](https://github.com/ros2/ros2/wiki/Linux-Development-Setup) and replace `release-latest` with `master` in the [get the ros2 code section](https://github.com/ros2/ros2/wiki/Linux-Development-Setup#get-ros-20-code). 
```
wget https://raw.githubusercontent.com/ros2/ros2/master/ros2.repos
```

Yingzhong262526, 2017-06-09:
Hi@mikaelarguedas ,
I do as you say! I create a new workspace my_ws by following  [these instructions](https://github.com/ros2/ros2/wiki/Linux-Development-Setup) and replace release-latest with master in the [get the ros2 code section](https://github.com/ros2/ros2/wiki/Linux-Development-Setup).Bridge communication between ros 1 and ros2 is installed and build as this link https://github.com/ros2/ros1_bridge/blob/master/README.md#build-the-bridge-from-source

After I install and build it,I try some example，I source my_ws/install/setup.bash and run talker  in a teminal and in another terminal I also source my_ws/install/setup.bash and run listerner ! It can works perfectly!

But when I want to test whether ros1 to ros2 brige can work ! as  [this instruction](https://github.com/ros2/ros1_bridge/blob/master/README.md#build-the-bridge-from-source).When I open shell D, 
run 
. /my_ws/install/setup.bash 
and next run
 listener

But there is nothing which does not listen the talker's message! Meanwhile in the shell B It also prints nothing.If correct, It should print "created 1to2 bridge for topic 'chatter' with ROS 1 type 'std_msgs/String' and ROS 2 type 'std_msgs/String'" .

You know that the dummy robot demo would use ros1_brige to communicate with ros2!

That is why? Is the the difference of ros2 version and ros1_brige package version?Sincerely expect your reply!

Karsten1987, 2017-06-09:
> build-the-bridge-from-source

We introduced namespaces lately and it seems as if the ros1_bridge does not
easily deal with it.
As a workaround, start your bridge with `dynamic_bridge
--bridge-all-2to1-topics`. This will force the publisher of ros2 topics
even if they are not literally matched with ROS1.

Yingzhong262526, 2017-06-09:
@Karsten1987 ,Thank you ! Now it works!

Yingzhong262526, 2017-06-09:
@mikaelarguedas ,
Now my problem is as follows:
In  the [dummy_robot demo instruction](https://github.com/ros2/ros2/wiki/dummy-robot-demo),step by step,after in  terminal 3,I run 

source ~/my_ws/install/setup.bash
python3 ~/my_ws/install/share/dummy_robot_bringup/launch/dummy_robot_bringup.py

Then in ternimal 2,the error occurs,

**terminate called after throwing an instance of 'std::length_error'
  what():  basic_string::_M_create
Aborted (core dumped)
**
How can I solve it?

Yingzhong262526, 2017-06-09:
@Karsten1987 ,
Hi,I do as you,in that example,ros1 bridge works! But in the [dummy_robot demo instruction](https://github.com/ros2/ros2/wiki/dummy-robot-demo),I meet that problem as above! Hope you can help me！

Karsten1987, 2017-07-18:
Hi @Yingzhong262526 
can you confirm that with the latest master oder beta3 this problem does still exist?
Which DDS are you using?

mikaelarguedas, 2017-08-08:
Assuming that the problem was fixed in beta2. @Yingzhong262526 feel free to comment here if it's not the case and we can reopen this issue

closed 2017-08-08 by mikaelarguedas