ros2/ros2 #359: "console_bridge" package building error

Yingzhong262526, 2017-06-13:
Hi:
  Everyone, I am installing ros2.0 on OS of  windows 10  building from source [as the tutorials](https://github.com/ros2/ros2/wiki/Windows-Development-Setup),But when I run and test use the command:
> python src\ament\ament_tools\scripts\ament.py test 
the errors occur and the buiding process stop,

+++ Building 'console_bridge'
==> 'C:\ros2\build\console_bridge\cmake__build.bat C:\Program Files\CMake\bin\cmake.EXE C:\ros2\src\vendor\console_bridge -DCMAKE_INSTALL_PREFIX=C:\ros2\install -G Visual Studio 14 2015 Win64' in 'C:\ros2\build\console_bridge'
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
CMake Error at CMakeLists.txt:2 (project):
  No CMAKE_C_COMPILER could be found.

CMake Error at CMakeLists.txt:2 (project):
  No CMAKE_CXX_COMPILER could be found.

-- Configuring incomplete, errors occurred!
See also "C:/ros2/build/console_bridge/CMakeFiles/CMakeOutput.log".
See also "C:/ros2/build/console_bridge/CMakeFiles/CMakeError.log".

<== Command 'C:\ros2\build\console_bridge\cmake__build.bat C:\Program Files\CMake\bin\cmake.EXE C:\ros2\src\vendor\console_bridge -DCMAKE_INSTALL_PREFIX=C:\ros2\install -G Visual Studio 14 2015 Win64' failed in 'C:\ros2\build\console_bridge' with exit code '1'
<== Command 'C:\ros2\build\console_bridge\cmake__build.bat C:\Program Files\CMake\bin\cmake.EXE C:\ros2\src\vendor\console_bridge -DCMAKE_INSTALL_PREFIX=C:\ros2\install -G Visual Studio 14 2015 Win64' failed in 'C:\ros2\build\console_bridge' with exit code '1'

Any ideas can help me ? Thanks very much!

wjwwood, 2017-06-13:
Looks like you're not running in a visual studio shell:

```
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
CMake Error at CMakeLists.txt:2 (project):
No CMAKE_C_COMPILER could be found.

CMake Error at CMakeLists.txt:2 (project):
No CMAKE_CXX_COMPILER could be found.

-- Configuring incomplete, errors occurred!
```

Yingzhong262526, 2017-06-13:
I run it in the vs2015 x64 native tools command prompt

mikaelarguedas, 2017-06-13:
The fact that the C++ compiler cannot be found indicates that something went wrong when setting up Visual Studio.
My guess would be that the Visual Studio C++ chackbox was not selected during the installation:
[image]

Detailed instructions to setup visual studio for the ROS2 project can be found on [this installation page](https://github.com/ros2/ros2/wiki/Windows-Install-Binary#install-visual-studio-community-2015).

Hope this helps

Yingzhong262526, 2017-06-13:
I can make sure that I select it in installation of VS 2015!

mikaelarguedas, 2017-07-19:
@Yingzhong262526 Can you confirm that this is fixed with the right VS2015 settings ?

mikaelarguedas, 2017-08-03:
WIthout reply I'm going to close this. @Yingzhong262526 please comment here if you face the issue again and we can reopen it

closed 2017-08-03 by mikaelarguedas

keimura, 2017-09-20:
Hi, I am attempting to install on windows and encountering the same error message. I have installed visual studios with the c++ checkbox checked as instructed.

keimura, 2017-09-21:
Please disregard my earlier comment. I re-did the installation tutorial and found that I was missing other dependencies.

cristipiticul, 2018-02-17:
For me, only [this](https://stackoverflow.com/a/48400271/1207267) worked (adding Windows 10 SDK in Visual Studio 14 2015)
To do that, you can restart the VS installation, and select "Modify". Under "Windows and Web Development" -> "Universal Windows App Development Tools", I checked
- Tools (1.4.1) and Windows 10 SDK
- Windows 10 SDK (other versions)

calvertdw, 2018-05-31:
I had the same issue. I relaunched the VS2015 installer and the C++ box was fully checked. I followed @cristipiticul's suggestion and the ament build started working. @mikaelarguedas can you update the documentation?

mikaelarguedas, 2018-05-31:
@calvertdw good to know this additional trouble shooting step worked for several people.

The wiki is editable by everyone, can you please submit an edit with the appropriate modification?