ros2/ros2 #360: Chocolatey install directory vs expected python location in ROS2

SecretaryBirds, 2017-06-13:
OS: Windows 10
problem command: `choco install -y python` and then any python script, like `topic_monitor.exe`.
output:
`Traceback (most recent call last):
  File "C:\Asi\master\ros2_windows\bin\topic_monitor-script.py", line 11, in <module>
    load_entry_point('topic-monitor==0.0.0', 'console_scripts', 'topic_monitor')()
  File "C:\ProgramData\chocolatey\lib\python3\tools\lib\site-packages\pkg_resources\__init__.py", line 549, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "C:\ProgramData\chocolatey\lib\python3\tools\lib\site-packages\pkg_resources\__init__.py", line 2542, in load_entry_point
    return ep.load()
  File "C:\ProgramData\chocolatey\lib\python3\tools\lib\site-packages\pkg_resources\__init__.py", line 2202, in load
    return self.resolve()
  File "C:\ProgramData\chocolatey\lib\python3\tools\lib\site-packages\pkg_resources\__init__.py", line 2208, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "C:\Asi\master\ros2_windows\Lib\site-packages\topic_monitor\scripts\topic_monitor.py", line 20, in <module>
    import rclpy
  File "C:\Asi\master\ros2_windows\Lib\site-packages\rclpy\__init__.py", line 18, in <module>
    from rclpy.impl.implementation_singleton import rclpy_implementation as _rclpy
  File "C:\Asi\master\ros2_windows\Lib\site-packages\rclpy\impl\implementation_singleton.py", line 32, in <module>
    rclpy_implementation = importlib.import_module('._rclpy', package='rclpy')
  File "C:\ProgramData\chocolatey\lib\python3\tools\lib\importlib\__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ImportError: Module use of python36.dll conflicts with this version of Python.`

When you use chocolatey to install, it doesn't install it to \ProgramData\chocolatey. It puts it at \Python36.

So, on systems where ROS2 binaries were built where python lives in the \ProgramData\chocolatey directory, they fail to run on system where python is somewhere else. Even with the PATH setup to find python in \Python36.

Either instructions need to change to help people force python to install in \ProgramData\chocolatey, or system needs to be more robust to using the python executable found in PATH, no matter where it is.

wjwwood, 2017-06-13:
I think this is a duplicate of https://github.com/ros2/ros2/issues/355#issuecomment-306735415, which was closed in favor of https://github.com/ros2/build_cop/issues/23.

We could update the instructions, but we're close to making a new release for beta 2, so it might be better to just update all the docs at once. @SecretaryBirds if you know what to change, go ahead and do it on the wiki (anyone can edit) and post here if you want us to review the changes.

dirk-thomas, 2017-06-13:
The important part is that we need to update the build nodes to install Python using the latest chocolatey version. That way users should have the same environment as the machine building the package.

mikaelarguedas, 2017-06-13:
Thanks for reporting, are you using the binaries from Beta-1 or the binaries from last night ?
Do you have the same issue if you set the `AMENT_PYTHON_PATH` environment variable to the location of where your python is installed?
We'll roll out new binaries for beta 2 in the next few days. Hopefully this issue will disappear with the building machine using the latest version of chocolatey packages. Though the fact that chocolatey now uses a folder named after the python version I'd expect the same issue to happen if you use a different version of python than the one used by the build machine

SecretaryBirds, 2017-06-13:
I'm running with a recent master build (probably a week or 2 old), and it's from a local/company build server. But the discrepancies in the Python paths are still an issue. Especially for those of us running Python3.5 (which choco installed in \ProgramData) and the build server, which is running with 3.6.
Apparently someone here placed 3.6 in \ProgramData also (probably manually).
So when I used choco to upgrade to 3.6 on my local machine, it put it in \Python36.

I'll look at the AMENT_PYTHON_PATH workaround.
We're also playing with just creating a symlink in Windows to point from \Python36 to \ProgamData\chocolatey for now.

mikaelarguedas, 2017-06-13:
yeah symlinking can be a good workaround. Trying out the ament_python_path is worth a shot and feedback would be greatly appreciated.

> Apparently someone here placed 3.6 in \ProgramData also (probably manually).

 Maybe you ended up with a manual python36 folder if someone tried the debug installation instructions (https://github.com/ros2/ros2/wiki/Windows-Development-Setup#extra-stuff-for-debug-mode)

SecretaryBirds, 2017-06-13:
Neither the AMENT_PYTHON_EXECUTABLE, or the AMEN_PYTHON_PATH variables are working for me.
local_setup.bat keeps failing with 
`'"C:\ProgramData\chocolatey\lib\python3\tools\python.exe"' is not recognized as an internal or external command,
operable program or batch file.`

Which is weird that it worked for @Yingzhong262526 in #355.

I'll just work around it manually and wait for beta2 to come out.
Maybe then, @wjwwood, we can think of an appropriate edit to the wiki.

SecretaryBirds, 2017-06-13:
Windows symlink works great.

`mklink /D \ProgramData\chocolatey\lib\python3\tools \Python36`

mikaelarguedas, 2017-07-19:
@SecretaryBirds  Can you confirm that the original issue has been fixed (with beta 2 or latest binaries)?

SecretaryBirds, 2017-07-19:
I tried the Windows binary, and it plain doesn't work for me. I downloaded the binary, extracted, sourced and ran: `ros2 run demo_nodes_py` and I get this error:

> Failed to load entry point 'start': DLL load failed: The operating system cannot run %1.
Failed to load entry point 'status': DLL load failed: The operating system cannot run %1.
Failed to load entry point 'stop': DLL load failed: The operating system cannot run %1.
Failed to load entry point 'list': DLL load failed: The operating system cannot run %1.
Failed to load entry point 'call': DLL load failed: The operating system cannot run %1.
Failed to load entry point 'list': DLL load failed: The operating system cannot run %1.
Failed to load entry point 'echo': DLL load failed: The operating system cannot run %1.
Failed to load entry point 'list': DLL load failed: The operating system cannot run %1.
Failed to load entry point 'pub': DLL load failed: The operating system cannot run %1.
Traceback (most recent call last):
  File "C:\Asi\beta2\ros2-windows\lib\demo_nodes_py\talker-script.py", line 11, in <module>
    load_entry_point('demo-nodes-py==0.0.0', 'console_scripts', 'talker')()
  File "C:\Python36\lib\site-packages\pkg_resources\__init__.py", line 561, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "C:\Python36\lib\site-packages\pkg_resources\__init__.py", line 2649, in load_entry_point
    return ep.load()
  File "C:\Python36\lib\site-packages\pkg_resources\__init__.py", line 2303, in load
    return self.resolve()
  File "C:\Python36\lib\site-packages\pkg_resources\__init__.py", line 2309, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "C:\Asi\beta2\ros2-windows\Lib\site-packages\topics\talker.py", line 18, in <module>
    import rclpy
  File "C:\Asi\beta2\ros2-windows\Lib\site-packages\rclpy\__init__.py", line 18, in <module>
    from rclpy.impl.implementation_singleton import rclpy_implementation as _rclpy
  File "C:\Asi\beta2\ros2-windows\Lib\site-packages\rclpy\impl\implementation_singleton.py", line 32, in <module>
    rclpy_implementation = importlib.import_module('._rclpy', package='rclpy')
  File "C:\Python36\lib\importlib\__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ImportError: DLL load failed: The operating system cannot run %1.

Interestingly enough, you can see that at least it's finding my python executable at C:\Python36, which was at the crux of this Issue.

SecretaryBirds, 2017-07-21:
I got R2B2 to compile against CoreDX using our internal Build Server, and when I run that on windows, the link is still important.

I don't get errors like above, but sourcing local_setup.bash fails to execute if the python link isn't in place.

When it is in place, everything works great.

This behavior could be a side-effect of the way our build server is setup and how it builds Ros2. I think we have our own fork of ament_cmake, but that's it. And it's only a few commits behind.

dirk-thomas, 2017-10-04:
Since the generated scripts in a release archive contain the path to the Python interpreter all we can do is make sure that we use the latest Python version installed by Chocolatey on our build machines and instruct the user to do the same. If either of them diverges it won't work. I don't see a way to workaround this.

In the meantime we have updated our build machines so the latest release should work fine if the user installed the same Python version.

Since I don't see anything else which we can do here I propose to close this ticket. I will do that for now. Please feel free to comment and this can be reopened if necessary.

closed 2017-10-04 by dirk-thomas

BrannonKing, 2017-10-04:
Suppose we transcribe the full path to python in the generated scripts. Just assume that it's in the path on Windows. I think that would be less problematic.

dirk-thomas, 2017-10-09:
> Just assume that it's in the path on Windows. I think that would be less problematic.

Do you know a way how we can influence the generation of the scripts to achieve this?