ros2/ros2 #361: improvements for topic_names_and_types

wjwwood, 2017-06-15:
This is an aggregate issue for this task.

Tasks:

- modify data structures to support multiple types per topic
- add get_service_names_and_types (and restrict get_topic_names_and_types to topics only)
- expose option to get topic names without demangling them (removing the ROS conventions using in the middleware)

Todo:

- [ ] finish connext static changes (still debugging some strange issues)
- [ ] make connext dynamic and opensplice compile again
- [ ] test get_service_names_and_types in rcl
- [ ] test no_demangle in rcl
- [ ] update test for avoid_ros_namespace_conventions to use no_demangle
- [ ] add get_sevice_names_and_types to rclcpp
- [ ] update bridge

Pr's for what I have already to come shortly.

wjwwood, 2017-06-15:
Just running some preliminary CI to catch issues I'm missing locally (no bridge and no connext yet):

- Linux:
  - [image]
- Linux-aarch64:
  - [image]
- macOS:
  - [image]
- Windows:
  - [image]

dirk-thomas, 2017-06-15:
I am not sure about the change to allow multiple different types per topic. In DDS it is clearly feasible. In ROS 1 it is not. Several "concepts" are based on this in ROS. E.g. when printing a message using `rostopic echo` you currently only specify the topic name. The same when e.g. visualizing data in `rqt_plot`, you pass `/topic/array_field[2]`. Those kind of interfaces need to change in order to support multiple different types per topic.

We should carefully consider how these are affected and need to be changed and what that means for migration / bridging before changing this.

wjwwood, 2017-06-15:
Since we don't have a central name service, we don't have an easy way to enforce topics have only one type. Until we come up with a way to do that, I think the information needs to be exposed, because it can (and so probably will) happen. It is useful to see from the tools that there is more than one type on the topic (indicates a problem).

Tools like echo, the bridge, etc. can just assert that only one type is associated with the topic and give up otherwise. In fact @mikaelarguedas and I were talking about that at lunch, and I have a pending change to my ros2cli pr that does that for echo. I'll make one for the bridge too, which just prints a warning if a topic has more than one type on it.

wjwwood, 2017-06-16:
Ok, I think the rest of these items can and should be ticketed for implementation later. Most are functions that aren't required for the cli or are just improvements to the tests. So I'm putting these all in review.

Here is CI:

- Linux:
  - [image]
- Linux-aarch64:
  - [image]
- macOS:
  - [image]
- Windows:
  - [image]

wjwwood, 2017-06-16:
The macOS warnings are related to https://github.com/ros2/rcutils/issues/32.

wjwwood, 2017-06-16:
@ros2/team bump on reviews for these. I need time to address comments and turn CI around before the cutoff. If you have time, say something here (to prevent multiple people from doing the review).

dirk-thomas, 2017-06-16:
I went through all the PRs. Due to the amount my "review" was likely more on the shallow side :wink: I made a few minor comments. Overall it looks good to me :+1:

wjwwood, 2017-06-16:
Thanks @dirk-thomas! I think I've addressed all the comments, here's new CI:

- Linux:
  - [image]
- Linux-aarch64:
  - [image]
- macOS:
  - [image]
- Windows:
  - [image]

And a Linux packaging job (to test the bridge):

- Linux packaging:
  - [image]

wjwwood, 2017-06-16:
Windows job got disconnected (restarted) and the Linux packaging job has a test failure. I'm looking into that now.

The macOS compiler warnings are expected for now.

wjwwood, 2017-06-16:
Ok, I fixed the bridge issue, here's one more CI (hopefully last):

- Linux:
  - [image]
- Linux-aarch64:
  - [image]
- Linux packaging:
  - [image]
- macOS:
  - [image]
- Windows:
  - [image]

I will merge if these look good.

wjwwood, 2017-06-17:
Windows test failure is known flakey.

wjwwood, 2017-06-17:
I've merged all the pr's and opened issues for all the leftover items I knew about.

closed 2017-06-17 by wjwwood