ros2/ros2 #362: Use CMAKE_X_STANDARD and check for compiler ID rather than platform

mikaelarguedas, 2017-06-16:
WIP, opening this for visibility:
common_interfaces: https://github.com/ros2/common_interfaces/commit/cfb32325ff9935b87765a14ef1ae0dc319265497
demos: https://github.com/ros2/demos/commit/d127266a115bb2c169bd34821cfe175fe839dce1
example_interfaces: https://github.com/ros2/example_interfaces/commit/9f769ea67da0d00d6659cbf74112c9dbc2e04a02
examples: https://github.com/ros2/examples/commit/24bc602dca905313a4162e1f7bbdab5fc84a9731
geometry2: https://github.com/ros2/geometry2/commit/c0d4f6971a64e2c8adba8cdde7ae87e402bd3b1d
rcl: https://github.com/ros2/rcl/compare/compile_options
rclcpp: https://github.com/ros2/rclcpp/commit/473c9244461cae9500f4df47caeb6b57d8af9c3f
rcl_interfaces: https://github.com/ros2/rcl_interfaces/commit/6b5d60716943da7b3a0d07ad0f93d7c5dc429e5c
realtime_support: https://github.com/ros2/realtime_support/commit/27396392e609f1c5942ea5532de892daf396259e
rmw: https://github.com/ros2/rmw/compare/compile_options
rmw_fastrtps: https://github.com/ros2/rmw_fastrtps/commit/98780e46238f34c857ca5abbf3c8a72409bf4577
rmw_opensplice: https://github.com/ros2/rmw_opensplice/commit/2255e2f663eaf423d4c9412a769d7b0568faa242
rmw_implementation: https://github.com/ros2/rmw_implementation/commit/94bb75b61a5817e8247e5db5a4e2e7d7b3df2d9f
rmw_connext: https://github.com/ros2/rmw_connext/commit/6c692ad08b39236c0b6ff6c8b0b934fb9c126f98
robot_model: https://github.com/ros2/robot_model/commit/7a1db8747520550db7fc142a762bd86c6b217da8
robot_state_publisher: https://github.com/ros2/robot_state_publisher/commit/8cd758043b6285927947e69b8681c47674cefe04
ros1_bridge: https://github.com/ros2/ros1_bridge/commit/d8cfc28770bf9af2766709c450262bb8c055c1b6
rosidl: https://github.com/ros2/rosidl/compare/compile_options
rosidl_typesupport: https://github.com/ros2/rosidl_typesupport/commit/91da499ed264e54986f37e09d96b3cf405d52f79
system_tests: https://github.com/ros2/system_tests/commit/122abd8213dffd40d98fdb0e32a0054d104954b6
urdfdom: https://github.com/ros2/urdfdom/commit/01037f82ac0a2a594f62353c546ee371b0e0f607

First round of CI:
* Linux [image]
* Linux-aarch64 [image]

mikaelarguedas, 2017-06-16:
@ros2/team CI is green for all the packages in the ros2.repos. CI on turtlebot repos almost done
No outstanding change except migrating every repo using a "std" flag to using CMAKE_X_STANDARD. Note that you can still find `-std=??` flags in the codebase when passing flags explicitely to specific targets.
We will need https://github.com/ros2/rcutils/pull/26 and https://github.com/ros2/rcl/pull/141 to be updated to finish the migration

dirk-thomas, 2017-06-16:
I reviewed the first block of changes and they all lgtm. I assume you will squash all changes into a single commit on the default branch? :+1:

mikaelarguedas, 2017-06-16:
yep, seems more convenient for everybody than opening PRs for all of them

mikaelarguedas, 2017-06-16:
:green_heart: :green_apple: :green_heart: on the :robot: repos file as well

dirk-thomas, 2017-06-16:
The generators can set the `C_STANDARD` on the target using this: http://docs.w3cub.com/cmake~3.5/prop_tgt/c_standard/#prop_tgt:C_STANDARD

mikaelarguedas, 2017-06-16:
Cool! I'll do that then!

wjwwood, 2017-06-16:
Belated, but the second block looks good.

mikaelarguedas, 2017-06-16:
:+1: thanks. I'll push those and open a PR for the rosidl one that may have a slightly different change

mikaelarguedas, 2017-06-17:
Ok all the commits landed. Not that all fils  that were not explicitely passing astandard glag to the compiler have not been modified. This issue trackd only replacing any `-std=c??` flags with setting the appropriate CMake standard variable

closed 2017-06-17 by mikaelarguedas