ros2/ros2 #365: Port rosserial to ROS 2

allenh1, 2017-06-19:
## Feature request

#### Feature description
Port `rosserial` libraries into ROS 2.

allenh1, 2017-08-04:
I'm going to close this -- I determined long ago this was not the correct way to do this.

mikaelarguedas, 2018-02-24:
@allenh1 Should the ros2/rosserial repository be deleted then?

allenh1, 2018-02-24:
@mikaelarguedas

closed 2018-02-24 by allenh1

ludusrusso, 2018-03-02:
Hi,
sorry for forcing discussion again here.. 

I'm experimenting ROS 2.0 and I'd like to have a standard way to connect small micro controllers (like Arduino UNO) with ROS2. Since you're not going to port the rosserial, which is the correct way to do so?

allenh1, 2018-03-02:
> I'm experimenting ROS 2.0 and I'd like to have a standard way to connect small micro controllers (like Arduino UNO) with ROS2. Since you're not going to port the rosserial, which is the correct way to do so?

This has been put into delay until I have some time to think about this, and we have some time to work things out better, but there's a couple of routes that come to mind.

The preferred route would be to get things building statically so that ROS 2 can run on an embedded device. This is a rather distant dream, however.

For now, I think the better route would be to get a micro DDS communication layer going for the arduino. Rosserial is not the correct route since it requires serialization of the data, which is less than desirable. At that point, it's best to just use ROS 1 over a bridge (in my opinion).

Some [work](https://github.com/PX4/Firmware/blob/4c4f990b3bff5e6bd92e2b282cb322a7da3ea075/msg/templates/urtps/microRTPS_transport.cpp) was done by the RTPS people to get microRTPS to be a part of the PX4 firmware. That might be a good place to look.

ludusrusso, 2018-03-02:
Thats sounds right! Anyway I think that rosserial is a perfect approach for super low power devices like the atmega of Arduino UNO! The bridge approach is perfect, but let just me understand why serialization is not a good approach!

gavanderhoorn, 2018-03-02:
@ludusrusso: one of those "micro DDS communication layer" that @allenh1 mentions could be [atolab/zhe](https://github.com/atolab/zhe). It's one of the proposed implementations of the XRCE DDS standard.

There's a bit of discussion on ROS Discourse about this.

Note: I haven't used this myself yet, so I don't know how well it would fit your use-case. Neither do I endorse this or any other XRCE proposal.

dirk-thomas, 2018-03-02:
Imo `rosserial` and `micro DDS` are no different from an architectural point of view. In both cases you are using a *different* protocol than what is being using in ROS 2 (aka RTPS / DDS). Therefore I think each user should choose what fits their needs the best. Of course it would be most desired to use a single protocol throughout your system but if that is not the case It doesn't really matter to which other protocols you bridge if you do bridge it in the first place.

allenh1, 2018-03-02:
> Of course it would be most desired to use a single protocol throughout your system but if that is not the case It doesn't really matter to which other protocols you bridge if you do bridge it in the first place.

I very much agree with that.

I wasn't meaning to imply that serialization is _bad_: it's going to happen at some point. You could have a simple, custom protocol that you write and serialize that ROS 2 would listen to.

The reason rosserial is not the answer, to me, is the fact that the code cannot be separated from ROS 1 in a way that isn't a complete rewrite of rosserial. I think that, in that situation, it's best to just write a new system.

ludusrusso, 2018-03-02:
@allenh1 that’s right! Anyway I like the rosserial idea, so, if for you is ok, I can work on a reimplementation of rosserial (let’s call it ros2serial) maintaining the original idea! So automatic library generation for targeted devices and api similar to rclcpp.

I’ll target Arduino but with the idea of extending it to other devices and platform!

allenh1, 2018-03-05:
@ludusrusso I'd defer to other members of the @ros2/team, but, as far as I'm concerned, you can go for it.

mikaelarguedas, 2018-03-07:
@ludusrusso I'm sure many users will find this valuable as I don't expect serial protocol to disappear from the embedded world anytime soon. Feel free to work on it. 

We also invite you to post on [discourse](https://discourse.ros.org/c/ng-ros) about your project, either early on to gather feedback on the design/implementation from like-minded people that are interested in the project or later when you have a working prototype and want to raise visibility and get feedback (it's up to you :smile: )

ludusrusso, 2018-03-07:
@mikaelarguedas I'll do it for sure :D thanks!!!

hemantku, 2018-08-15:
@ludusrusso:  Have you been able to port this as ros2serial?

vmayoral, 2019-01-23:
You may want to have a look at OFERA European project (https://microros.github.io/) as an alternative to rosserial for ROS 2.

flynneva, 2020-09-17:
for anyone who is interested i started porting the rosserial_python package over to ros2.

https://github.com/flynneva/ros2serial_python

still a lot of work to do but its a start.